OAID/Tengine #1: Failed to build.

blaueck, 2018-01-02:
# Error message:
`
gcc  -std=gnu99 -mcpu=cortex-a72 -Wa,-mcpu=cortex-a72 -Wl,-mcpu=cortex-a72 -g    -c /home/abc/tengine/build/executor/operator/arm/fc/sgemv_1x8.s -o /home/abc/tengine/build/executor/operator/arm/fc/sgemv_1x8.o
sgemv_1x8.S: Assembler messages:
sgemv_1x8.S:83: Error: bad instruction `ldp x9,x10,[sp]'
sgemv_1x8.S:84: Error: bad instruction `ldp x11,x12,[sp,0x10]'
sgemv_1x8.S:85: Error: bad instruction `ldp x13,x14,[sp,0x20]'
sgemv_1x8.S:86: Error: ARM register expected -- `ldr x15,[sp,0x30]'
sgemv_1x8.S:88: Error: bad instruction `ldp q30,q31,[x0]'
sgemv_1x8.S:90: Error: selected processor does not support `cbz x15,loop_end' in ARM mode
sgemv_1x8.S:94: Error: ARM register expected -- `ldr q0,[x1],#0x10'
sgemv_1x8.S:95: Error: bad instruction `ld4 { v16.d,v17.d,v18.d,v19.d}[0],[x2],#0x20'
sgemv_1x8.S:96: Error: bad instruction `ld4 { v16.d,v17.d,v18.d,v19.d}[1],[x3],#0x20'
sgemv_1x8.S:97: Error: bad instruction `ld4 { v20.d,v21.d,v22.d,v23.d}[0],[x4],#0x20'
sgemv_1x8.S:98: Error: bad instruction `ld4 { v20.d,v21.d,v22.d,v23.d}[1],[x5],#0x20'
sgemv_1x8.S:100: Error: bad instruction `fmla v30.4s,v16.4s,v0.s[0]'
sgemv_1x8.S:101: Error: bad instruction `fmla v30.4s,v17.4s,v0.s[1]'
sgemv_1x8.S:102: Error: ARM register expected -- `subs x15,x15,0x1'
sgemv_1x8.S:103: Error: bad instruction `fmla v30.4s,v18.4s,v0.s[2]'
sgemv_1x8.S:104: Error: bad instruction `prfm pldl1keep,[x1,0x80]'
sgemv_1x8.S:105: Error: bad instruction `fmla v30.4s,v19.4s,v0.s[3]'
sgemv_1x8.S:106: Error: bad instruction `prfm pldl1keep,[x2,0xa0]'
sgemv_1x8.S:108: Error: bad instruction `fmla v31.4s,v20.4s,v0.s[0]'
sgemv_1x8.S:109: Error: bad instruction `prfm pldl1keep,[x3,0x90]'
sgemv_1x8.S:110: Error: bad instruction `fmla v31.4s,v21.4s,v0.s[1]'
sgemv_1x8.S:111: Error: bad instruction `prfm pldl1keep,[x4,0xb0]'
sgemv_1x8.S:112: Error: bad instruction `fmla v31.4s,v22.4s,v0.s[2]'
sgemv_1x8.S:113: Error: bad instruction `prfm pldl1keep,[x5,0xc0]'
sgemv_1x8.S:114: Error: bad instruction `fmla v31.4s,v23.4s,v0.s[3]'
sgemv_1x8.S:116: Error: unexpected character `n' in type specifier
sgemv_1x8.S:116: Error: bad instruction `b.ne loop'
sgemv_1x8.S:122: Error: bad instruction `st1 {v30.s}[0],[x6]'
sgemv_1x8.S:123: Error: bad instruction `st1 {v30.s}[1],[x7]'
sgemv_1x8.S:124: Error: bad instruction `st1 {v30.s}[2],[x9]'
sgemv_1x8.S:125: Error: bad instruction `st1 {v30.s}[3],[x10]'
sgemv_1x8.S:126: Error: bad instruction `st1 {v31.s}[0],[x11]'
sgemv_1x8.S:127: Error: bad instruction `st1 {v31.s}[1],[x12]'
sgemv_1x8.S:128: Error: bad instruction `st1 {v31.s}[2],[x13]'
sgemv_1x8.S:129: Error: bad instruction `st1 {v31.s}[3],[x14]'
sgemv_1x8.S:131: Error: bad instruction `ret'
/home/abc/tengine/scripts/makefile.build:105: recipe for target '/home/abc/tengine/build/executor/operator/arm/fc/sgemv_1x8.o' failed
`

cyberfire, 2018-01-02:
@blaueck 

Could you please post your gcc spec by  gcc -v?
I guess it is probably your gcc is built for aarch32

Thanks,

Haitao

blaueck, 2018-01-02:
@cyberfire , thanks for you quick reply. Here is the output by gcc -v.
# Output by gcc -v
`
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/5/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.5' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --enable-multilib --disable-sjlj-exceptions --with-arch=armv7-a --with-fpu=vfpv3-d16 --with-float=hard --with-mode=thumb --disable-werror --enable-multilib --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.5)
`

cyberfire, 2018-01-02:
@blaueck 

As you can see from the log, your gcc is built for armv7-a, which is aarch32.
Current we do not support armv7 yet.

Thanks,

Haitao

blaueck, 2018-01-02:
Ok, I got it. I will try to find an aarch64 version of gcc. @cyberfire , Thank you for your help.

closed 2018-01-02 by blaueck