OAID/Tengine #4: Performance compared with ncnn or others

knsong, 2018-01-10:
Hi, 
First, thank you all for the work. But are there some more benchmarks compared with other highly optimized embeded inference engine such as [ncnn](https://github.com/Tencent/ncnn) and [In-Prestissimo](https://github.com/in66-dev/In-Prestissimo)?

cyberfire, 2018-01-11:
@knsong 

We do not have time to do that yet.

It will be great that if you could do some apple to apple benchmark and let us know:-)

Thanks,

Haitao

knsong, 2018-01-22:
@cyberfire  I simply wrote a arm 32-bit version of `ConvFast` as Tengine, and compared it with `ncnn`'s direct convolution on mobilenet. The performance is ~300ms or so on Nexus 5x, while ncnn's is ~600ms. That's great. But on smaller networks whose input is small such as 60x60x3, the overhead in `im2col` is too heavy compared with the gemm part, which results in a inferior performance compared with ncnn.

00liujj, 2018-02-28:
@knsong would you share your arm 32-bit implementation details? the 4x16 version of sgemm needs a lot of aarch64 registers which armv7a doesn't have.

knsong, 2018-02-28:
@00liujj  For armv7a, I rewrote a 4x4 version of gemm for `ConvFast` instead of 4x16 using registers `q0~q11`.

00liujj, 2018-02-28:
@knsong Thank you for your comments. Is it possible to implement depthwise conv by gemm?

knsong, 2018-02-28:
@00liujj  yes, but the performance may be not such good as direct convolution.

00liujj, 2018-03-01:
@knsong Thank you very much!

knsong, 2018-03-01:
@00liujj Interleave means that both the rows of kernel and the colums of inputs are interleaved. This is a common gemm optimization trick(loop unrolling to use simd acceleration and make it cache friendly)  on cpu. Maybe [reading here](https://github.com/flame/how-to-optimize-gemm) can further help you.

00liujj, 2018-03-01:
@knsong thank you. I can now understand the code faster after your info.

closed 2018-03-09 by knsong