OAID/Tengine #7: what's the suitable version of protoc ?

zhonghuabao1, 2018-03-06:
Hello! I get error while `make`:
```
make[1]: Entering directory '/home/pi/tengine/serializer'
make[2]: Entering directory '/home/pi/tengine/serializer/onnx'
protoc --cpp_out . onnx.proto
mv onnx.pb.h ../include
In file included from onnx.pb.cc:4:0:
/home/pi/tengine/serializer/include/onnx.pb.h:12:2: error: #error This file was generated by a newer version of protoc which is
 #error This file was generated by a newer version of protoc which is
  ^~~~~
/home/pi/tengine/serializer/include/onnx.pb.h:13:2: error: #error incompatible with your Protocol Buffer headers. Please update
 #error incompatible with your Protocol Buffer headers.  Please update
  ^~~~~
/home/pi/tengine/serializer/include/onnx.pb.h:14:2: error: #error your headers.
 #error your headers.

```
It seen that I using the wrong version of protoc , isn't it ?

zhonghuabao1, 2018-03-06:
have deal it !
I use 3.5.1 before, now libprotoc 3.0.0 will be ok!

cyberfire, 2018-03-08:
@zhonghuabao1 
You can use "make clean" to clear the generated .pb.h and .pb.cc files.

Thanks,

Haitao

zhonghuabao1, 2018-03-09:
@cyberfire “make clean” is not used when I use protoc3.5.1
I had try it !

closed 2018-04-11 by cyberfire