OAID/Tengine #8: Get error when compile batch_norm layer !

zhonghuabao1, 2018-03-06:
```
g++ -std=c++11   -I. -I./include  -O3 -funroll-loops  -Wall -g -fPIC  -I/home/pi/tengine/executor/../core/include -I/home/pi/tengine/executor/../include -I/home/pi/tengine/executor/../operator/include  -I/home/pi/tengine/executor/include   -DCONFIG_ARCH_ARM64=1   -c batch_norm.cpp -o /home/pi/tengine/build/executor/operator/arm/batch_norm.o 
batch_norm.cpp: In member function ‘virtual bool TEngine::BatchNormImpl::BNOps::Prerun(TEngine::Node*)’:
batch_norm.cpp:79:29: error: ‘sqrt’ is not a member of ‘std’
        scale_var_inv[c]=1.f/std::sqrt(var[c]*rescale_factor + eps);
                             ^~~
/home/pi/tengine/scripts/makefile.build:97: recipe for target '/home/pi/tengine/build/executor/operator/arm/batch_norm.o' failed
make[3]: *** [/home/pi/tengine/build/executor/operator/arm/batch_norm.o] Error 1
make[3]: Leaving directory '/home/pi/tengine/executor/operator/arm'
/home/pi/tengine/scripts/makefile.build:70: recipe for target 'arm' failed
make[2]: *** [arm] Error 2
make[2]: Leaving directory '/home/pi/tengine/executor/operator'
Makefile:72: recipe for target 'operator' failed
make[1]: *** [operator] Error 2
make[1]: Leaving directory '/home/pi/tengine/executor'
Makefile:52: recipe for target 'executor' failed
make: *** [executor] Error 2
```
I have not idea to this!

zhonghuabao1, 2018-03-06:
have deal it!
modify `tengine/executor/operator/arm/batch_norm.cpp`:
add `include<math.h> `on the head
find `std::sqrt` and delete `std::`

cyberfire, 2018-03-08:
@zhonghuabao1 
Thank you for the reporting on compiling issue. I guess it is probably related with g++ version. 
Could you please post the g++ version here?

Thanks,

Haitao

zhonghuabao1, 2018-03-15:
@cyberfire 
my g++ version is 6.3.0
my 64-bit OS use this: `https://github.com/bamarni/pi64`

cyberfire, 2018-04-11:
Thanks for the feedback.

-Haitao

closed 2018-04-11 by cyberfire