OAID/Tengine #18: build error on nvidia tx2

dongxiao92, 2018-06-21:
I try to build Tengine on Nvidia TX2 .But I got the following errors.

/usr/lib/aarch64-linux-gnu/mesa/libGL.so.1: undefined reference to `drmCloseOnce'
/usr/lib/aarch64-linux-gnu/mesa/libGL.so.1: undefined reference to `drmMap'
/usr/lib/aarch64-linux-gnu/mesa/libGL.so.1: undefined reference to `drmUnmap'
/usr/lib/aarch64-linux-gnu/mesa/libGL.so.1: undefined reference to `drmOpenOnce'

The makefile.config is listed as follow.
# Set the target arch
CONFIG_ARCH_ARM64=y

# Enable Compiling Optimization
CONFIG_OPT_CFLAGS = -O2

# uncomment to build the reference operators wrapped on Caffe
CONFIG_CAFFE_REF=y

# Set the path of caffe used for compile and link,
# if caffe's operator calculation is enabled.
CAFFE_ROOT = /home/nvidia/dongxiao/caffe-1.0

# Enable GPU support by Arm Computing Library
#CONFIG_ACL_GPU=y
#

# Use BLAS as the operator implementation
#
# CONFIG_ARCH_BLAS=y

# Set the path of ACL
#ACL_ROOT=/home/nvidia/dongxiao/ComputeLibrary

# Enable other serializers
CONFIG_CAFFE_SUPPORT=y
# CONFIG_MXNET_SUPPORT=y
# CONFIG_ONNX_SUPPORT=y
# CONFIG_TF_SUPPORT=y

# Enable Wrappers
# CONFIG_FRAMEWORK_WRAPPER=y

The OS is ubuntu16.04 and the gcc used is gcc (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.9).

dongxiao92, 2018-06-21:
I solved that problem by change the libGL.so used to link.
But when I try to run the bench_sqz，I got the segmentatation fault.
./build/tests/bin/bench_sqz -r1
tensor: prob created by node: prob is not consumed
add the node: prob into output list
Load model successfully
[1] 25556 segmentation fault (core dumped)

cyberfire, 2018-06-21:
@dongxiao92 
Are you doing cross-compiling or native build?
A quick suggestion is to comment out those two lines in makefile.config
CAFFE_ROOT = /home/nvidia/dongxiao/caffe-1.0
CONFIG_CAFFE_SUPPORT=y

Thanks,
Haitao

dongxiao92, 2018-06-22:
@cyberfire 
I'm not doing cross-compiling.  I build Tengine on Tx2 natively.
The build error above is due to that the libGL.so under /usr/lib/aarch64-linux-gnu/mesa/ seems to have some problems and I have solved it by linking to another libGL.so.
But now the segmentation failt error occured.

cyberfire, 2018-06-25:
@dongxiao92 
How is going?  As I do not have TX2 board on hand, I could not debug the issue  now. 
Did you get the issue resolved? or could you please use gdb to run the program and dump the stack ?

Thanks,

Haitao

lyuchuny3, 2018-12-07:
Hi, dongxiao92,
Have you try the lastest version of Tengine? Did your problem solved?

dongxiao92, 2018-12-07:
@lyuchuny3 Hi, I try the current master branch and it can be compiled. But the test seems to have problems. I run the ./build/tests/bin/bench_sqz -r10.  The calling to 'load_model' at line162 failed.

cyberfire, 2018-12-12:
@dongxiao92 
It looks that the makefile.config is not the copy of the newest one (makefile.config.example).
So, could you please build and try again with the new makefile.config?

Thanks,

Haitao

dongxiao92, 2018-12-12:
@cyberfire Thanks for your reply.
I tried using the newest makefile.config and then re-compiled. The tests encountered a new problem.
The makefile.config and how to reproduce the problem is listed below.

> \#-------------------------------------------------------------------------------
> \#  Template configuration for compiling
> \#
> \#  First copy makefile.config.example to makefile.config (assume you are on the 
> \#  root directory). 
> \#
> \#  $ cp makefile.config.example makefile.config
> \#
> \#  Next modify makefile.config, and then compile by
> \#
> \#  $ make
> \#
> \#  or build in parallel with 8 threads
> \#
> \#  $ make -j8 
> \#-------------------------------------------------------------------------------
> 
> \# Set the target arch 
> CONFIG_ARCH_ARM64=y
> 
> \# Enable Compiling Optimization
> CONFIG_OPT_CFLAGS = -O2 
> 
> \# Enable GPU support by Arm Computing Library
> \# CONFIG_ACL_GPU=y
> \#
>  
> 
> \# Use BLAS as the operator implementation
> \#
> CONFIG_ARCH_BLAS=y
> 
> \# Set the path of ACL 
> \# ACL_ROOT=/home/firefly/ComputeLibrary
> 
> \# Enable other serializers 
> CONFIG_CAFFE_SERIALIZER=y
> \# CONFIG_MXNET_SERIALIZER=y
> CONFIG_ONNX_SERIALIZER=y
> \# CONFIG_TF_SERIALIZER=y
> CONFIG_TENGINE_SERIALIZER=y
> 
> \# Enable Wrappers
> \# CONFIG_FRAMEWORK_WRAPPER=y
>
> \# version postfix
> CONFIG_VERSION_POSTFIX=github

To inform the openblas pasth to linker, I make the following modification in makefile:
>ifeq ($(CONFIG_ARCH_BLAS),y)
>    LIB_LDFLAGS+=-lopenblas -L/home/nvidia/dongxiao/libs/OpenBLAS-0.2.20/build/lib/
>endif
then,
> make -j8
> ./build/tests/bin/bench_sqz -r10

The error is:
tensor: prob created by node: prob is not consumed
add the node: prob into output list
Load model successfully
*** Error in `./build/tests/bin/bench_sqz': free(): invalid next size (fast): 0x0000000000527cf0 ***
[1]    2673 abort (core dumped)  ./build/tests/bin/bench_sqz -r10

The environment information possibly related is listed below:

> gcc --version

gcc (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> uname -a 

Linux tegra-ubuntu 4.4.38-tegra #1 SMP PREEMPT Thu Mar 1 20:49:20 PST 2018 aarch64 aarch64 aarch64 GNU/Linux

dongxiao92, 2018-12-13:
@cyberfire 
I locate a out-of-bound bug in the cpu_probe.cpp, which leads to the segmentfault in my situation. But I'm not clear about the logic of current code. Thus, I think sending a PR now is not a good idea. Do you have any suggestions about the next movement?

dongxiao92, 2018-12-13:
close as solved.

closed 2018-12-13 by dongxiao92