OAID/Tengine #19: make Raspberry_pi3 for bug

molyswu, 2018-06-23:
Hi, 

My hardware platform uses Raspberry_pi3 (A53)，compiled to which is the following error: 
sgemv_1x8_a72.S:Assebler messages: 
sgemv_1x8_a72.S:73 Error: ARM register expected ..'cmp x3,0x4'
sgemv_1x8_a72.S:74 Error:bad instruction'prfm pldlkeep,[x1,0x40]' 
sgemv_1x8_a72.S:75 Error:bad instruction'prfm pldlkeep,[x2,0x200]' 
sgemv_1x8_a72.S:76 Error:bad instruction'prfm pldlkeep,[x2,0x240]' 

thanks for your help!

cyberfire, 2018-06-25:
@molyswu 
I guess you are probably running an AARCH32 Linux, instead of AARCH64.  
Could you please post the output of 'gcc -v' here ?

Also, maybe you can edit makefile.config:
disable CONFIG_ARCH_ARM64 and enable CONFIG_ARCH_BLAS

You need a blas library in PI3 board to build it.  It can be done in Ubuntu by this command:
apt install libopenblas-dev

Thanks,

Haitao

molyswu, 2018-06-27:
my gcc-v :4.9.2

molyswu, 2018-06-27:
The experimental 64 bit operating system of aspberry Pi 3

cyberfire, 2018-06-27:
@molyswu 

Could you please put all the output of 'gcc -v' here ?

Thanks,

Haitao

cyberfire, 2018-08-30:
Close. Since Tengine can build and run on Raspberry now.

closed 2018-08-30 by cyberfire