OAID/Tengine #24: firefly rk3399 board

wmorders, 2018-07-01:
i build this project on firefly rk3399 board, and run the benchmark .
The result is always as follows :
    tensor:prob created by node: prob is not consumed
    add the node : prob into output list
    load model successfully
    ..........
    0.2763 - "n02123045 tabby, tabby cat"
    0.2673 - "n02123159 tiger cat"
    0.1766 - "n02119789 kit fox, Vulpes macrotis"
	0.0827 - "n02124075 Egyptian cat"
    0.0777 - "n02085620 Chihuahua"
the parameter do not take effect ; and What should I do?  thanks!

cyberfire, 2018-07-02:
@wmorders 

I guess you run build/tests/bin/bench_sqz, is that correct?

What kind of parameters you are going to set?  To set another picture to test?   If so, you have to revise the tests/bin/bench_sqz.cpp to change the picture.

Thanks,

Haitao

wmorders, 2018-07-03:
I run  build/tests/bin/bench_sqz –p 4  ;  build/tests/bin/bench_sqz –p 4,5   ;  build/tests/bin/bench_sqz –p 0  and build/tests/bin/bench_sqz –p 0,1,2,3  after  set performance policy Correctly .

wmorders, 2018-07-03:
And the result is always as follows :
tensor:prob created by node: prob is not consumed
add the node : prob into output list
load model successfully
..........
0.2763 - "n02123045 tabby, tabby cat"
0.2673 - "n02123159 tiger cat"
0.1766 - "n02119789 kit fox, Vulpes macrotis"
0.0827 - "n02124075 Egyptian cat"
0.0777 - "n02085620 Chihuahua"

closed 2018-07-04 by wmorders