OAID/Tengine #25: Build Error with CONFIG_ACL_GPU=y in RasbperryPi3 B

ghimiredhikura, 2018-07-01:
# Hardware: RaspberryPi3 Model B

this is how I build compute library
$ scons Werror=1 debug=0 asserts=0 neon=0 opencl=1 examples=1 os=linux arch=armv7a -j3

# makefile.config 

```
#-------------------------------------------------------------------------------
#  Template configuration for compiling
#
#  First copy makefile.config.example to makefile.config (assume you are on the 
#  root directory). 
#
#  $ cp makefile.config.example makefile.config
#
#  Next modify makefile.config, and then compile by
#
#  $ make
#
#  or build in parallel with 8 threads
#
#  $ make -j8
#-------------------------------------------------------------------------------

# Set the target arch 
#CONFIG_ARCH_ARM64=y

# Enable Compiling Optimization
CONFIG_OPT_CFLAGS = -O2

# uncomment to build the reference operators wrapped on Caffe
# CONFIG_CAFFE_REF=y

# Set the path of caffe used for compile and link,
# if caffe's operator calculation is enabled.
# CAFFE_ROOT = /home/firefly/caffe

# Enable GPU support by Arm Computing Library
CONFIG_ACL_GPU=y
#

# Use BLAS as the operator implementation
#
CONFIG_ARCH_BLAS=y

# Set the path of ACL 
ACL_ROOT=/home/pi/Desktop/project/ComputeLibrary

# Enable other serializers 
CONFIG_CAFFE_SUPPORT=y
# CONFIG_MXNET_SUPPORT=y
# CONFIG_ONNX_SUPPORT=y
# CONFIG_TF_SUPPORT=y

# Enable Wrappers
# CONFIG_FRAMEWORK_WRAPPER=y
```

# tengine build error
```
cc1plus: all warnings being treated as errors
/home/pi/Desktop/project/tengine/scripts/makefile.build:108: recipe for target '/home/pi/Desktop/project/tengine/build/driver/acl_graph/acl_graph_driver.o' failed
make[2]: *** [/home/pi/Desktop/project/tengine/build/driver/acl_graph/acl_graph_driver.o] Error 1
make[2]: *** Waiting for unfinished jobs....
g++ -std=c++11    -I.  -O2 -Wall -g -fPIC  -I/home/pi/Desktop/project/tengine/driver/include  -I/home/pi/Desktop/project/tengine/driver/../include  -I/home/pi/Desktop/project/tengine/driver/../core/include -I/home/pi/Desktop/project/tengine/driver/../operator/include  -I/home/pi/Desktop/project/tengine/driver/../executor/include  -Werror -DCONFIG_ACL_GPU=1  -I/home/pi/Desktop/project/ComputeLibrary -I/home/pi/Desktop/project/ComputeLibrary/include -I/home/pi/Desktop/project/ComputeLibrary/utils -c acl_conv.cpp -o /home/pi/Desktop/project/tengine/build/driver/acl/acl_conv.o 
cc1plus: all warnings being treated as errors
/home/pi/Desktop/project/tengine/scripts/makefile.build:108: recipe for target '/home/pi/Desktop/project/tengine/build/driver/acl/acl_driver.o' failed
make[2]: *** [/home/pi/Desktop/project/tengine/build/driver/acl/acl_driver.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[... 247 lines cut ...]
/home/pi/Desktop/project/ComputeLibrary/arm_compute/core/CL/OpenCL.h:121:5: note: in expansion of macro ‘DECLARE_FUNCTION_PTR’
     DECLARE_FUNCTION_PTR(clEnqueueSVMUnmap);
     ^~~~~~~~~~~~~~~~~~~~
/home/pi/Desktop/project/ComputeLibrary/arm_compute/core/CL/OpenCL.h:79:38: error: ignoring attributes on template argument ‘cl_int(cl_command_queue, _cl_event**) {aka int(_cl_command_queue*, _cl_event**)}’ [-Werror=ignored-attributes]
     std::function<decltype(func_name)> func_name##_ptr = nullptr
                                      ^
/home/pi/Desktop/project/ComputeLibrary/arm_compute/core/CL/OpenCL.h:122:5: note: in expansion of macro ‘DECLARE_FUNCTION_PTR’
     DECLARE_FUNCTION_PTR(clEnqueueMarker);
     ^~~~~~~~~~~~~~~~~~~~
/home/pi/Desktop/project/ComputeLibrary/arm_compute/core/CL/OpenCL.h:79:38: error: ignoring attributes on template argument ‘cl_int(cl_uint, _cl_event* const*) {aka int(unsigned int, _cl_event* const*)}’ [-Werror=ignored-attributes]
     std::function<decltype(func_name)> func_name##_ptr = nullptr
                                      ^
/home/pi/Desktop/project/ComputeLibrary/arm_compute/core/CL/OpenCL.h:123:5: note: in expansion of macro ‘DECLARE_FUNCTION_PTR’
     DECLARE_FUNCTION_PTR(clWaitForEvents);
     ^~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
/home/pi/Desktop/project/tengine/scripts/makefile.build:108: recipe for target '/home/pi/Desktop/project/tengine/build/driver/acl/acl_conv.o' failed
make[2]: *** [/home/pi/Desktop/project/tengine/build/driver/acl/acl_conv.o] Error 1
make[2]: Leaving directory '/home/pi/Desktop/project/tengine/driver/acl'
/home/pi/Desktop/project/tengine/scripts/makefile.build:72: recipe for target 'acl' failed
make[1]: *** [acl] Error 2
make[1]: Leaving directory '/home/pi/Desktop/project/tengine/driver'
Makefile:131: recipe for target 'driver' failed
make: *** [driver] Error 2
```

ghimiredhikura, 2018-07-02:
# build tengine driver without -Werror option. 

It seems that those build error are due to -Werror (warning error) option in tengine driver Makefile. 
Therefore I rebuild tengine without -Werror option, and build was successful. 

```
driver-$(CONFIG_CAFFE_REF)+=caffe/
driver-$(CONFIG_ACL_GPU)+=acl_graph/
driver-$(CONFIG_ACL_GPU)+=acl/

obj-y+=cpu/ plugin/  $(driver-y)

INC_DIR+=-I$(shell pwd)/include 
INC_DIR+=-I$(shell pwd)/../include 
INC_DIR+=-I$(shell pwd)/../core/include
INC_DIR+=-I$(shell pwd)/../operator/include 
INC_DIR+=-I$(shell pwd)/../executor/include 

CXXFLAGS+=

COMMON_CFLAGS+=$(CONFIG_OPT_CFLAGS)
#COMMON_CFLAGS+= -Wall -g -fPIC  $(INC_DIR) -Werror
COMMON_CFLAGS+= -Wall -g -fPIC  $(INC_DIR)

ifeq ($(CONFIG_CAFFE_REF),y)
    COMMON_CFLAGS+= -DCONFIG_CAFFE_REF=1
endif

ifeq ($(CONFIG_ARCH_ARM64),y)
    COMMON_CFLAGS+= -DCONFIG_ARCH_ARM64=1
endif

ifeq ($(CONFIG_ACL_GPU),y)
    COMMON_CFLAGS+= -DCONFIG_ACL_GPU=1
endif

ifeq ($(CONFIG_ARCH_ARM32),y)
    COMMON_CFLAGS+= -DCONFIG_ARCH_ARM32=1
endif

install:
```

# but gor cl::Error runtime in testing 
$ ./build/tests/bin/bench_sqz -d acl_opencl

Output
```
pi@raspberrypi:~/Desktop/project/tengine $ ./build/tests/bin/bench_sqz -d acl_opencl
terminate called after throwing an instance of 'cl::Error'
  what():  empty
Aborted
```

cyberfire, 2018-07-04:
@ghimiredhikura 

Just a double check, do you use the Arm Compute library v18.05?

Thanks,

Haitao

ghimiredhikura, 2018-07-08:
Yes I used v18.05 version. And I used openblas library. 
It will be nice, if you can check in Pi3 device. 
Btw, is there any plan to support neon instructions from ACL library. I think currently it only supports opencl from ACL. 

Thanks,

Deepak

closed 2018-11-25 by cyberfire