OAID/Tengine #27: failed compilation with hand-compiled libprotobuf.so(2.6.1)

peyer, 2018-07-11:
@HEGALLIS I have tried to compiled protobuf(2.6.1) and put the path  libprotobuf.so  to android_build_armv8.sh, while when Linking lintengine.so, the compilation info shown that generated_message_reflection.h:496: undefined reference to `typeinfo for google::protobuf::Message'

the te_caffe.pb.cc file was generated by host protoc(x86_64),

cyberfire, 2018-07-12:
@peyer 

Is the version of protoc in your x86_64 is 2.61  too? 
Also, besides libprotobuf.so, please overwrite the headers files too.

Thanks,

Haitao

lyuchuny3, 2018-07-31:
try to build with protobuf 3.0

peyer, 2018-08-13:
@cyberfire @lyuchuny3 @HEGALLIS . Sorry about reply you so late. I have successfully compiled protobuf 3.0 on x86_64 and ARM. And I changed Makefile to compile Tengine on x86_64 to transform caffemodel to tengine format on host, it worked now. Thanks again!