OAID/Tengine #29: where TEngine::RegisterConv2dOpencl() defined?

wmorders, 2018-07-13:
where TEngine::RegisterConv2dOpencl() definition and implementation ?

cyberfire, 2018-07-16:
@wmorders 

Just simply remove the directory build/ and rebuild the whole project. 
There are still some issues in make system and the fastest solution is to build from scratch after change the makefile.config.

Thanks,

Haitao

cyberfire, 2018-07-24:
Fixed.

closed 2018-07-24 by cyberfire

wmorders, 2018-07-24:
I  build from scratch after change the makefile.config. 
the build log:
make[2]: Leaving directory '/home/tim/workspace/Tengine/tools/bin'
aarch64-linux-gnu-g++ --sysroot=/home/tim/workspace/Tengine/sysroot/ubuntu_rootfs  -L/usr/lib/aarch64-linux-gnu -L/lib/aarch64-linux-gnu  /home/tim/workspace/Tengine/build/tools/bin/test_accuracy.o  -o /home/tim/workspace/Tengine/build/tools/bin/test_accuracy -L../build -ltengine -lprotobuf -ldl -lpthread -lresolv -L/usr/lib/aarch64-linux-gnu -lopencv_calib3d -lopencv_contrib -lopencv_core -lopencv_features2d -lopencv_flann -lopencv_gpu -lopencv_highgui -lopencv_imgproc -lopencv_legacy -lopencv_ml -lopencv_objdetect -lopencv_ocl -lopencv_photo -lopencv_stitching -lopencv_superres -lopencv_ts -lopencv_video -lopencv_videostab -Wl,-rpath=/usr/lib/aarch64-linux-gnu/mesa  -L../build
../build/libtengine.so: undefined reference to `TEngine::RegisterConv2dOpencl()'
collect2: error: ld returned 1 exit status
Makefile:66: recipe for target '/home/tim/workspace/Tengine/build/tools/bin/test_accuracy' failed
make[1]: *** [/home/tim/workspace/Tengine/build/tools/bin/test_accuracy] Error 1
make[1]: Leaving directory '/home/tim/workspace/Tengine/tools'
Makefile:142: recipe for target 'tools' failed

wmorders, 2018-07-24:
can u tell me where TEngine::RegisterConv2dOpencl() definition and implementation ?
i find the whole project source code , but i can not find the definition and implementation.
Thanks

lyuchuny3, 2018-07-25:
This is because you change your makefile.config and you haven't "make clean". Try `make clean` / `rm -rf build` ,and rebuild `make`. This problem should be solved.

TEngine::RegisterConv2dOpencl()   is in  `executor/plugin/init.hpp`
its implement lies in  `driver/acl/acl_conv.cpp`