OAID/Tengine #30: Is depthwise convolution supported?

nopattern, 2018-07-25:
Hi，I want to import mobilefacenet into Tengine.  The  mobilefacnet uses a depthwise convolution .  And the mxnet load also failed(mxnet json file uploaded).  The error is as followings:
                   

> Parse the json file error in line 11
Parse the json file error in line 17
Parse the json file error in line 23
Parse the json file error in line 24
Parse the json file error in line 25
Parse the json file error in line 26
Parse the json file error in line 27
Parse the json file error in line 28
Parse the json file error in line 29
Parse the json file error in line 31
Parse the json file error in line 32

[model-symbol.json.txt](https://github.com/OAID/Tengine/files/2227177/model-symbol.json.txt)

lyuchuny3, 2018-07-25:
1. depthwise convolution is supported
2.  for  the mxnet load problem, I would like to know you run the tengine code on which platform (e.g. rk3399)?

nopattern, 2018-07-26:
yes, my test platform is rk3399.

nopattern, 2018-07-26:
Pull the new version and test good. Thank you.

nopattern, 2018-07-27:
Hi,@jing-openailab 
     I try to use the prerun_graph, run_graph, and postrun_graph as in test_mxnet_mobilenet.cpp. The first time is good while the second time failed.  I have uploaded the changed version of test_mxnet_mobilenet.cpp. Do I misunderstand something?
[test_mxnet_mobilenet.cpp.txt](https://github.com/OAID/Tengine/files/2234823/test_mxnet_mobilenet.cpp.txt)

lyuchuny3, 2018-07-30:
try to replace my `main` function in `test/bin/test_mxnet_mobilenet.cpp`, build, run `./build/tests/bin/test_mxnet_mobilenent -r3`, results should be right each time.

``` 
int main(int argc, char * argv[])
{
	
   int res;
   // load labels
   std::vector<std::string> labels;
   LoadLabelFile(labels,label_file);

   while((res=getopt(argc,argv,"r:"))!=-1)
   {
      switch(res)
      {
         case 'r':
            repeat_count=strtoul(optarg,NULL,10);
            break;
         default:
            break;
      }
   }

   const char * model_name="mobilenet";

   // 1. init tengine lib
   init_tengine_library();
   if(request_tengine_version("0.1")<0)
       return 1;

    // 2. load model
   if(load_model(model_name,"mxnet",text_file,model_file)<0)
       return 1; 
   std::cout<<"Load model successfully\n";

    // 3. create graph
   graph_t graph=create_runtime_graph("graph0",model_name,NULL);

    // 4. allocate input data, set input shape && prerun
   int img_h=224;
   int img_w=224;
   float  * input_data=(float*) malloc (sizeof(float) * img_h *img_w *3);
   int node_idx = 0;
   int tensor_idx = 0;
   tensor_t input_tensor=get_graph_input_tensor(graph , node_idx , tensor_idx );
   tensor_t output_tensor=get_graph_output_tensor(graph, node_idx, tensor_idx);
   int dims[]={1,3,img_h,img_w};
   set_tensor_shape(input_tensor,dims,4);
   prerun_graph(graph);

  
   // 5. run, each time set input buffer
  
   for(int i=0;i<repeat_count;i++)
   {
        /* set input buffer */
        get_input_data( image_file, input_data, img_h, img_w, channel_mean,0.017);
        if(set_tensor_buffer(input_tensor,input_data,3*img_h*img_w*4)<0)
        {
            std::printf("Set buffer for tensor failed\n");
            return -1;
        }
        run_graph(graph,1);

        /* get output */
        
        int count=get_tensor_buffer_size(output_tensor)/4;
        float *  data=(float *)(get_tensor_buffer(output_tensor));
        float * end=data+count;
        std::vector<float> result(data, end);
        std::vector<int> top_N=Argmax(result,5);
        
        for(unsigned int i=0;i<top_N.size();i++)
        {
            int idx=top_N[i];

            std::cout<<std::fixed << std::setprecision(4)<<result[idx]<<" - \"";
            std::cout<< labels[idx]<<"\"\n";
        }
        printf("=========\n");
        
   }

   // 6. post run
   put_graph_tensor(input_tensor);
   put_graph_tensor(output_tensor);

   postrun_graph(graph);  

   destroy_runtime_graph(graph);
   remove_model(model_name);
	
   free(input_data);

   std::cout<<"ALL TEST DONE\n";

   release_tengine_library();
   return 0;
}
```

nopattern, 2018-07-30:
Thanks again.

closed 2018-07-30 by nopattern