OAID/Tengine #31: RRi3B+ build failed.

ghimiredhikura, 2018-07-31:
I used following setting in the makefile.config. 

'''
#-------------------------------------------------------------------------------
#  Template configuration for compiling
#
#  First copy makefile.config.example to makefile.config (assume you are on the 
#  root directory). 
#
#  $ cp makefile.config.example makefile.config
#
#  Next modify makefile.config, and then compile by
#
#  $ make
#
#  or build in parallel with 8 threads
#
#  $ make -j8
#-------------------------------------------------------------------------------

# Set the target arch 
#CONFIG_ARCH_ARM64=y

# Enable Compiling Optimization
CONFIG_OPT_CFLAGS = -O2

# uncomment to build the reference operators wrapped on Caffe
# CONFIG_CAFFE_REF=y

# Set the path of caffe used for compile and link,
# if caffe's operator calculation is enabled.
# CAFFE_ROOT = /home/firefly/caffe

# Enable GPU support by Arm Computing Library
# CONFIG_ACL_GPU=y
#

# Use BLAS as the operator implementation
#
CONFIG_ARCH_BLAS=y

# Set the path of ACL 
# ACL_ROOT=/home/firefly/ComputeLibrary

# Enable other serializers 
CONFIG_CAFFE_SERIALIZER=y
# CONFIG_MXNET_SERIALIZER=y
# CONFIG_ONNX_SERIALIZER=y
# CONFIG_TF_SERIALIZER=y
CONFIG_TENGINE_SERIALIZER=y

# Enable Wrappers
# CONFIG_FRAMEWORK_WRAPPER=y
'''

but while building the library i got following error. any help please? 

'''
g++    /home/pi/Desktop/Tengine/build/tools/bin/test_accuracy.o  -o /home/pi/Desktop/Tengine/build/tools/bin/test_accuracy -Wl,-rpath,/home/pi/Desktop/Tengine/tools/../build -ltengine -lprotobuf -ldl -lpthread -lresolv -L/usr/local/lib -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_dnn_objdetect -lopencv_dpm -lopencv_face -lopencv_photo -lopencv_freetype -lopencv_fuzzy -lopencv_hfs -lopencv_img_hash -lopencv_line_descriptor -lopencv_optflow -lopencv_reg -lopencv_rgbd -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_surface_matching -lopencv_tracking -lopencv_datasets -lopencv_text -lopencv_dnn -lopencv_plot -lopencv_xfeatures2d -lopencv_shape -lopencv_video -lopencv_ml -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_highgui -lopencv_videoio -lopencv_flann -lopencv_xobjdetect -lopencv_imgcodecs -lopencv_objdetect -lopencv_xphoto -lopencv_imgproc -lopencv_core -Wl,-rpath=/usr/lib/aarch64-linux-gnu/mesa  -L/home/pi/Desktop/Tengine/tools/../build
/usr/bin/ld: /home/pi/Desktop/Tengine/build/tools/bin/test_accuracy: hidden symbol `_ZN6google8protobuf8internal17AssignDescriptorsERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEPKNS1_15MigrationSchemaEPKPKNS0_7MessageEPKjPNS0_14MessageFactoryEPNS0_8MetadataEPPKNS0_14EnumDescriptorEPPKNS0_17ServiceDescriptorE' in /usr/local/lib/libprotobuf.a(generated_message_reflection.cc.o) is referenced by DSO
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
Makefile:66: recipe for target '/home/pi/Desktop/Tengine/build/tools/bin/test_accuracy' failed
make[1]: *** [/home/pi/Desktop/Tengine/build/tools/bin/test_accuracy] Error 1
make[1]: Leaving directory '/home/pi/Desktop/Tengine/tools'
Makefile:145: recipe for target 'tools' failed
make: *** [tools] Error 2
'''

lyuchuny3, 2018-08-02:
This problem  may because, there are two version of protobuf. Try to rm all versions of protobuf, and install one

ghimiredhikura, 2018-08-06:
Thank you. Solved.

closed 2018-08-06 by ghimiredhikura