OAID/Tengine #33: RaspberryPi3B+ build example error

ghimiredhikura, 2018-08-04:
```
pi@raspberrypi:~/Desktop/Tengine_new/examples/benchmark/build $ cmake ..
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenCV: /usr/local (found version "3.4.2") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Desktop/Tengine_new/examples/benchmark/build
pi@raspberrypi:~/Desktop/Tengine_new/examples/benchmark/build $ make
Scanning dependencies of target test_convolution
[ 12%] Building C object CMakeFiles/test_convolution.dir/test_convolution.c.o
[ 25%] Linking C executable test_convolution
/home/pi/Desktop/Tengine_new/examples/benchmark/../../install/lib/libtengine.so: undefined reference to `TEngine::RegisterBatchNormNodeExec()'
/home/pi/Desktop/Tengine_new/examples/benchmark/../../install/lib/libtengine.so: undefined reference to `TEngine::RegisterConv2dDepth()'
/home/pi/Desktop/Tengine_new/examples/benchmark/../../install/lib/libtengine.so: undefined reference to `TEngine::RegisterFullyConnectedFast()'
/home/pi/Desktop/Tengine_new/examples/benchmark/../../install/lib/libtengine.so: undefined reference to `TEngine::RegisterConv2dFast()'
/home/pi/Desktop/Tengine_new/examples/benchmark/../../install/lib/libtengine.so: undefined reference to `TEngine::RegisterPoolingNodeExec()'
/home/pi/Desktop/Tengine_new/examples/benchmark/../../install/lib/libtengine.so: undefined reference to `TEngine::RegisterScaleNodeExec()'
collect2: error: ld returned 1 exit status
CMakeFiles/test_convolution.dir/build.make:94: recipe for target 'test_convolution' failed
make[2]: *** [test_convolution] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/test_convolution.dir/all' failed
make[1]: *** [CMakeFiles/test_convolution.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

cyberfire, 2018-08-05:
@ghimiredhikura 
Would you please dump the makefile.config here?

Thanks,

haitao

ghimiredhikura, 2018-08-05:
CONFIG_ARCH_ARM32=y

everything else exactly same as in makefile.config.example

lyuchuny3, 2018-08-06:
This github version only support arm64 assembly codes.  We do open this option `CONFIG_ARCH_ARM32=y`. If you want to run Tengine on arm32, you can config this way:

```
# CONFIG_ARCH_ARM64=y
CONFIG_ARCH_BLAS=y
```
, that means close ` CONFIG_ARCH_ARM64` and open ` CONFIG_ARCH_BLAS`, everything else exactly same as in makefile.config.example。

ghimiredhikura, 2018-08-06:
Yes, I can successfully use Tengine with BLAS on arm32. Do you have plan to support arm32 with CONFIG_ARCH_ARM32=y. If yes, will there be any difference in inference speed using ARCH_BLAS or ARCH_ARM32? Thnaks.

gaojinwei01, 2018-08-06:
Hi ghimiredhikura,

We have Armv7 version NN library which is faster than BLAS. We do have plan to release it in the future.

Xiaowei

ghimiredhikura, 2018-08-12:
Hello gaxjinwei01 

Please let me know if you release this Armv7 version of NN library. 
Thanks

Deepak

closed 2018-12-07 by lyuchuny3