OAID/Tengine #36: fatal error: arm_neon.h: No such file or directory

gunjupark, 2018-08-17:
I built ACL(ComputingLabrary) by
`scons Werror=1 -j4 debug=0 asserts=1 neon=0 opencl=1 embed_kernels=1 os=linux arch=arm64-v8a`
command.

After that , in Tengine directory
vi makefile.config

`# Set the target arch 
 19 CONFIG_ARCH_ARM64=y
 20 
 21 # Enable Compiling Optimization
 22 CONFIG_OPT_CFLAGS = -O2
 23 
 24 # uncomment to build the reference operators wrapped on Caffe
 25 # CONFIG_CAFFE_REF=y
 26 
 27 # Set the path of caffe used for compile and link,
 28 # if caffe's operator calculation is enabled.
 29 # CAFFE_ROOT = /home/firefly/caffe
 30 
 31 # Enable GPU support by Arm Computing Library
 32  CONFIG_ACL_GPU=y
 33 #
 34 
 35 
 36 # Use BLAS as the operator implementation
 37 #
 38  CONFIG_ARCH_BLAS=y
 39 
 40 # Set the path of ACL 
 41  ACL_ROOT=/home/admin/workspace/ComputeLibrary
 42 
 43 # Enable other serializers 
 44 CONFIG_CAFFE_SERIALIZER=y
 45 # CONFIG_MXNET_SERIALIZER=y
 46 # CONFIG_ONNX_SERIALIZER=y
 47 # CONFIG_TF_SERIALIZER=y
 48 CONFIG_TENGINE_SERIALIZER=y
 49 
 50 # Enable Wrappers
 51 # CONFIG_FRAMEWORK_WRAPPER=y
`

make -j8
then occured these errors

`make[2]: Entering directory '/home/admin/workspace/Tengine/driver/acl_graph'
g++ -std=c++11    -I.  -Wno-ignored-attributes -Werror -O2 -Wall -g -fPIC  -I/home/admin/workspace/Tengine/driver/include  -I/home/admin/workspace/Tengine/driver/../include  -I/home/admin/workspace/Tengine/driver/../core/include -I/home/admin/workspace/Tengine/driver/../operator/include  -I/home/admin/workspace/Tengine/driver/../executor/include   -DCONFIG_ARCH_ARM64=1 -DCONFIG_ACL_GPU=1  -I/home/admin/workspace/ComputeLibrary -I/home/admin/workspace/ComputeLibrary/include -I/home/admin/workspace/ComputeLibrary -c acl_graph_driver.cpp -o /home/admin/workspace/Tengine/build/driver/acl_graph/acl_graph_driver.o 
**In file included from acl_graph_device.hpp:35:0,
                 from acl_graph_driver.hpp:36,
                 from acl_graph_driver.cpp:25:**
acl_graph.hpp:31:22: fatal error: arm_neon.h: No such file or directory
compilation terminated.
`

How can I solve this problem??

gunjupark, 2018-08-17:
oh I have mistake
gcc & g++ version not correct in this Makefile
sorry~

closed 2018-08-17 by gunjupark

pangchaogit, 2019-03-27:
I got same error ,I use cmake cpmile ,how do  I set g++ in Cmakelists.txt