OAID/Tengine #37: libopenblas.so, needed by /home/admin/workspace/Tengine/install/lib/libtengine.so, not found (try using -rpath or -rpath-link)

gunjupark, 2018-08-17:
```
-- Check for working C compiler: /home/admin/tmp/android-ndk-r16b/toolchains/llvm/prebuilt/linux-x86_64/bin/clang
-- Check for working C compiler: /home/admin/tmp/android-ndk-r16b/toolchains/llvm/prebuilt/linux-x86_64/bin/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/admin/tmp/android-ndk-r16b/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++
-- Check for working CXX compiler: /home/admin/tmp/android-ndk-r16b/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_TOOLCHAIN_FILE = /home/admin/tmp/android-ndk-r16b/build/cmake/android.toolchain.cmake
-- CMAKE_INSTALL_PREFIX = /usr/local
-- Found OpenCV: /home/admin/tmp/OpenCV-android-sdk (found version "3.4.2") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/admin/workspace/Tengine/examples/build
```

Cmake-version : 3.12.1
Protobuf-version: 3.0.0
OpenCV-sdk-version : 3.4.2

Tengine/examples/android-build-armv8.sh  : 
```
#!/bin/bash

export ANDROID_NDK=/home/admin/tmp/android-ndk-r16b

cmake -DCMAKE_TOOLCHAIN_FILE=$ANDROID_NDK/build/cmake/android.toolchain.cmake \
     -DANDROID_ABI="arm64-v8a" \
     -DANDROID_PLATFORM=android-21 \
     -DANDROID_STL=gnustl_shared \
     -DTENGINE_DIR=/home/admin/workspace/Tengine \
     -DOpenCV_DIR=/home/admin/tmp/OpenCV-android-sdk/sdk/native/jni \
     -DPROTOBUF_DIR=/home/admin/tmp/protobuf_lib \
     ..  
```

Tengine/android-build-armv8.sh  : 
```
#!/bin/bash

PROTOBUF_PATH=
BLAS_PATH=
ARCH_TYPE=
ANDROID_NDK=

while read line 
do
   IFS=:
   arr=($line)
   if [ "${arr[0]}" == "PROTOBUF_DIR" ]; then
      PROTOBUF_PATH=${arr[1]}
   elif [ "${arr[0]}" == "BLAS_DIR" ]; then
      BLAS_PATH=${arr[1]}
   elif [ "${arr[0]}" == "CONFIG_ARCH_TYPE" ]; then
      ARCH_TYPE=${arr[1]}
   elif [ "${arr[0]}" == "ANDROID_NDK" ]; then
      ANDROID_NDK=${arr[1]}
   fi  
done<../android_config.txt

cmake -DCMAKE_TOOLCHAIN_FILE=$ANDROID_NDK/build/cmake/android.toolchain.cmake \
    -DANDROID_ABI="arm64-v8a" \
    -DCONFIG_ARCH_BLAS=ON \
    -DANDROID_PLATFORM=android-21 \
    -DANDROID_STL=gnustl_shared \
[... 95 lines cut ...]
make[2]: *** [faster_rcnn/FASTER_RCNN] Error 1
CMakeFiles/Makefile2:255: recipe for target 'faster_rcnn/CMakeFiles/FASTER_RCNN.dir/all' failed
make[1]: *** [faster_rcnn/CMakeFiles/FASTER_RCNN.dir/all] Error 2
/home/admin/tmp/android-ndk-r16b/toolchains/aarch64-linux-android-4.9/prebuilt/linux-x86_64/lib/gcc/aarch64-linux-android/4.9.x/../../../../aarch64-linux-android/bin/ld: warning: libopenblas.so, needed by /home/admin/workspace/Tengine/install/lib/libtengine.so, not found (try using -rpath or -rpath-link)
/home/admin/workspace/Tengine/install/lib/libtengine.so: undefined reference to `cblas_sgemm'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
imagenet_classification/CMakeFiles/Classify.dir/build.make:151: recipe for target 'imagenet_classification/Classify' failed
make[2]: *** [imagenet_classification/Classify] Error 1
CMakeFiles/Makefile2:365: recipe for target 'imagenet_classification/CMakeFiles/Classify.dir/all' failed
make[1]: *** [imagenet_classification/CMakeFiles/Classify.dir/all] Error 2
/home/admin/workspace/Tengine/install/lib/libtengine.so: undefined reference to `cblas_sgemm'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
mtcnn/CMakeFiles/MTCNN.dir/build.make:166: recipe for target 'mtcnn/MTCNN' failed
make[2]: *** [mtcnn/MTCNN] Error 1
CMakeFiles/Makefile2:90: recipe for target 'mtcnn/CMakeFiles/MTCNN.dir/all' failed
make[1]: *** [mtcnn/CMakeFiles/MTCNN.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

I don't guess reason that error occured.

How can I solve this error??? 

help me

gunjupark, 2018-08-17:
soooory 
I missed `-DBLAS_DIR ` in Tengine/examples/android-build-armv8.sh
thanks

closed 2018-08-17 by gunjupark

dzyjjpy, 2019-03-19:
how do you solve the issue? I met the same issue.
 in Tengine/examples/android-build-armv8.sh, add -DBLAS_DIR

android-build-armv8.sh content: 
#!/bin/bash

export ANDROID_NDK=/home/jiapy/workspace/tengine/Tengine_android_build/android-ndk-r16b

cmake -DCMAKE_TOOLCHAIN_FILE=$ANDROID_NDK/build/cmake/android.toolchain.cmake \
     -DANDROID_ABI="arm64-v8a" \
     -DANDROID_PLATFORM=android-21 \
     -DANDROID_STL=c++_shared \
     -DTENGINE_DIR=/home/jiapy/workspace/tengine \
     -DOpenCV_DIR=/home/jiapy/workspace/tengine/Tengine_android_build/opencv/sdk/native/jni\
     -DPROTOBUF_DIR=/home/jiapy/workspace/tengine/Tengine_android_build/protobuf_lib \
     -DBLAS_DIR=/home/jiapy/workspace/tengine/Tengine_android_build/Openblas020-android \
     ..

Error information:
~/workspace/tengine/examples/build$ make -j8
Scanning dependencies of target SSD
Scanning dependencies of target LIGHTEN_CNN
Scanning dependencies of target Classify
Scanning dependencies of target convert_caffe_to_tm
Scanning dependencies of target FASTER_RCNN
Scanning dependencies of target MTCNN
Scanning dependencies of target YOLOV2
Scanning dependencies of target MSSD
[  3%] Building CXX object tengine_model/convert/CMakeFiles/convert_caffe_to_tm.dir/convert_caffe_to_tm.cpp.o
[  6%] Building CXX object mobilenet_ssd/CMakeFiles/MSSD.dir/mssd.cpp.o
[ 10%] Building CXX object lighten_cnn/CMakeFiles/LIGHTEN_CNN.dir/lighten_cnn.cpp.o
[ 13%] Building CXX object ssd/CMakeFiles/SSD.dir/ssd.cpp.o
[ 20%] Building CXX object imagenet_classification/CMakeFiles/Classify.dir/classification.cpp.o
[ 20%] Building CXX object yolov2/CMakeFiles/YOLOV2.dir/yolov2.cpp.o
[ 23%] Building CXX object faster_rcnn/CMakeFiles/FASTER_RCNN.dir/faster_rcnn.cpp.o
[ 26%] Building CXX object mtcnn/CMakeFiles/MTCNN.dir/mtcnn_utils.cpp.o
[ 30%] Building CXX object tengine_model/convert/CMakeFiles/convert_caffe_to_tm.dir/__/__/common/common.cpp.o
[ 33%] Building CXX object lighten_cnn/CMakeFiles/LIGHTEN_CNN.dir/__/common/common.cpp.o
[ 36%] Linking CXX executable convert_caffe_to_tm
/home/jiapy/workspace/tengine/Tengine_android_build/Openblas020-android/arm64/lib/libopenblas.so: undefined reference to `stderr'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
tengine_model/convert/CMakeFiles/convert_caffe_to_tm.dir/build.make:100: recipe for target 'tengine_model/convert/convert_caffe_to_tm' failed
make[2]: *** [tengine_model/convert/convert_caffe_to_tm] Error 1
CMakeFiles/Makefile2:491: recipe for target 'tengine_model/convert/CMakeFiles/convert_caffe_to_tm.dir/all' failed
make[1]: *** [tengine_model/convert/CMakeFiles/convert_caffe_to_tm.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 40%] Building CXX object mtcnn/CMakeFiles/MTCNN.dir/mtcnn.cpp.o
[ 43%] Building CXX object mtcnn/CMakeFiles/MTCNN.dir/test_mtcnn.cpp.o
[ 46%] Building CXX object ssd/CMakeFiles/SSD.dir/__/common/common.cpp.o
[ 50%] Building CXX object mobilenet_ssd/CMakeFiles/MSSD.dir/__/common/common.cpp.o
[ 53%] Building CXX object yolov2/CMakeFiles/YOLOV2.dir/__/common/common.cpp.o
[ 56%] Linking CXX executable LIGHTEN_CNN
[ 60%] Building CXX object faster_rcnn/CMakeFiles/FASTER_RCNN.dir/__/common/common.cpp.o
[ 63%] Linking CXX executable SSD
[ 66%] Building CXX object imagenet_classification/CMakeFiles/Classify.dir/model_config.cpp.o
/home/jiapy/workspace/tengine/Tengine_android_build/Openblas020-android/arm64/lib/libopenblas.so: undefined reference to `stderr'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
lighten_cnn/CMakeFiles/LIGHTEN_CNN.dir/build.make:137: recipe for target 'lighten_cnn/LIGHTEN_CNN' failed
make[2]: *** [lighten_cnn/LIGHTEN_CNN] Error 1
CMakeFiles/Makefile2:305: recipe for target 'lighten_cnn/CMakeFiles/LIGHTEN_CNN.dir/all' failed
make[1]: *** [lighten_cnn/CMakeFiles/LIGHTEN_CNN.dir/all] Error 2
[ 70%] Building CXX object imagenet_classification/CMakeFiles/Classify.dir/__/common/common.cpp.o
[ 73%] Linking CXX executable MSSD
[ 76%] Linking CXX executable YOLOV2
[ 80%] Building CXX object mtcnn/CMakeFiles/MTCNN.dir/__/common/common.cpp.o
[ 83%] Linking CXX executable FASTER_RCNN
[ 86%] Linking CXX executable Classify
[ 90%] Linking CXX executable MTCNN
/home/jiapy/workspace/tengine/Tengine_android_build/Openblas020-android/arm64/lib/libopenblas.so: undefined reference to `stderr'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
[... 11 lines cut ...]
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
yolov2/CMakeFiles/YOLOV2.dir/build.make:137: recipe for target 'yolov2/YOLOV2' failed
make[2]: *** [yolov2/YOLOV2] Error 1
CMakeFiles/Makefile2:140: recipe for target 'yolov2/CMakeFiles/YOLOV2.dir/all' failed
make[1]: *** [yolov2/CMakeFiles/YOLOV2.dir/all] Error 2
/home/jiapy/workspace/tengine/Tengine_android_build/Openblas020-android/arm64/lib/libopenblas.so: undefined reference to `stderr'
/home/jiapy/workspace/tengine/Tengine_android_build/Openblas020-android/arm64/lib/libopenblas.so: undefined reference to `stderr'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
faster_rcnn/CMakeFiles/FASTER_RCNN.dir/build.make:137: recipe for target 'faster_rcnn/FASTER_RCNN' failed
make[2]: *** [faster_rcnn/FASTER_RCNN] Error 1
CMakeFiles/Makefile2:250: recipe for target 'faster_rcnn/CMakeFiles/FASTER_RCNN.dir/all' failed
make[1]: *** [faster_rcnn/CMakeFiles/FASTER_RCNN.dir/all] Error 2
[... cut ...]
/home/jiapy/workspace/tengine/Tengine_android_build/Openblas020-android/arm64/lib/libopenblas.so: undefined reference to `stderr'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
mtcnn/CMakeFiles/MTCNN.dir/build.make:167: recipe for target 'mtcnn/MTCNN' failed
make[2]: *** [mtcnn/MTCNN] Error 1
CMakeFiles/Makefile2:85: recipe for target 'mtcnn/CMakeFiles/MTCNN.dir/all' failed
make[1]: *** [mtcnn/CMakeFiles/MTCNN.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2