OAID/Tengine #40: Cannot find load function for operator: Reshape

pby5, 2018-08-24:
I use mobilenet025 which download from mxnet gluon model zoos, and run test_mxnet_mobilenet.cpp ,it occurs problem"Cannot find load function for operator: Reshape",  but I find Tengine supports "Reshape" operater,.

pby5, 2018-08-24:
where can I download other mxnet mobilenet models with different params(0.25) like your mxnet mobilenet models? Thanks

jing-openailab, 2018-08-27:
The mobilenet mxnet model in out tengine model zoo is converted from the caffe model by using the tool that mxnet provides, because mobilenet mxnet model was not available in mxnet model zoo last year.
Would you please send us your mobilenet025 mxnet model, and we'll see why Tengine fails.

pby5, 2018-08-28:
can you leave a email and I post the models to you?Thanks

jing-openailab, 2018-08-28:
jingyou@openailab.com

pby5, 2018-08-28:
models has been posted to your emails

lyuchuny3, 2018-08-28:
Hi, master had updated.  Parsing Reshape in Mxnet model supported now. You can try your model.