OAID/Tengine #42: parse file: ./models/ssd_mobilenet.pb failed

betsyweilin, 2018-08-31:
Is there any requirement of pb files? I use other pb files like download from mobilenet ,or trained by myself , when run test_tf_mobilenet, it shows:
parse file: ./models/ssd_mobilenet.pb failed
Release workspace default resource

So I wonder if there is any special requirement?

lyuchuny3, 2018-08-31:
1. Firstly, you can downlowd  `frozen_mobilenet_v1_224.pb` from [Tengine model Zoo](https://pan.baidu.com/share/init?surl=LXZ8vOdyOo50IXS0CUPp8g) with  passwd( 57vb), and then run `tests/bin/test_tf_mobilenet`, it should be successfully parse the pb file.
2. For your problem of parsing  `ssd_mobilenet.pb`, could you give me a link for your model with random params? 
Thanks

hey-yahei, 2018-10-17:
1. The model you trained is modelnet ssd, but `test_tf_mobilenet` is to test [mobilenet](https://github.com/tensorflow/models/tree/master/research/slim) model.
2. For the [ssd_mobilenet_v1 model provided by research/object_detection](https://github.com/tensorflow/models/blob/master/research/object_detection/g3doc/detection_model_zoo.md), Tengine does not support all  operations about preprocess and postprocess currently. 

    Training your own mobilenet_ssd in caffe with [chuanqi305's model](https://github.com/chuanqi305/MobileNet-SSD) is suggested.

    If you insist on using it, try to modify the network and remove all operations about preprocess and postprocess which can be replaced by C++ code in your application. Just like the model implemented by chuanqi305 with caffe and the Tengine example [mobilenet_ssd](https://github.com/OAID/Tengine/blob/master/examples/mobilenet_ssd/mssd.cpp).

closed 2019-03-02 by cyberfire