OAID/Tengine #43: parse ssd_mobilenet_v1.pb failed

moneyloveyang, 2018-09-04:
I trained my ssd_mobillenet in tensorflow, then use models/research/object_detection/export_inference_graph.py freeze .ckpt file to get  ssd_mobilenet_v1.pb. When I load and parse ssd_mobilenet.pb failed. The error happened in function TFSerializer::ConstructGraph(). It looks like some TFNode's inputs'name are not match with TFNode name, so TFNode can not find its input nodes then set up connection failed.There are two examples：
   1）TFNode‘s name is ’Preprocessor/map/TensorArray‘ ，but parse his output TFNode's input, we just find 'Preprocessor/map/TensorArray:1'
  2) TFNode‘s name is ’Preprocessor/map/while/Identity‘ ，but parse his output TFNode's input, we just find '^Preprocessor/map/while/Identity'.

hey-yahei, 2018-09-17:
For the names of the TFNode, you can remove the ":0" and "^" from tensor names by modifying the `TFSerializer::ConstructGraph()` function, but this is only a **temporary solution**. 

In addition, for the [ssd_mobilenet_v1 model provided by models/research/object_detection](https://github.com/tensorflow/models/blob/master/research/object_detection/g3doc/detection_model_zoo.md), Tengine does not support all  operations about preprocess and postprocess currently. 
If you insist on using it, try to modify the network and remove all operations about preprocess and postprocess which can be replaced by C++ code in your application. Just like [the model implemented by chuanqi305 with caffe](https://github.com/chuanqi305/MobileNet-SSD) and the Tengine example [mobilenet_ssd](https://github.com/OAID/Tengine/blob/master/examples/mobilenet_ssd/mssd.cpp).

gongdalinux, 2019-04-30:
@moneyloveyang Do you have solved this problem? Any example?
@hey-yahei modify the network? Can you give more details?
Thank you !

cyberfire, 2019-07-17:
@moneyloveyang 

please follow the tflite object detection guide to convert the trained model into a fused pb or a tflite model.

Thanks,

Haitao

closed 2019-07-17 by cyberfire