OAID/Tengine #45: Build error!

clhne, 2018-09-05:
sgemv_1x8_a72.S: Assembler messages:
sgemv_1x8_a72.S:67: Error: alignment not a power of 2
sgemv_1x8_a72.S:73: Error: too many memory references for `cmp'
sgemv_1x8_a72.S:74: Error: no such instruction: `prfm pldl1keep,[x1,0x40]'
sgemv_1x8_a72.S:75: Error: no such instruction: `prfm pldl1keep,[x2,0x200]'
sgemv_1x8_a72.S:76: Error: no such instruction: `prfm pldl1keep,[x2,0x240]'
sgemv_1x8_a72.S:77: Error: no such instruction: `movi d30,0'
sgemv_1x8_a72.S:78: Error: no such instruction: `movi d31,0'
sgemv_1x8_a72.S:79: Error: no such instruction: `cbz x0,start_convolution'
sgemv_1x8_a72.S:80: Error: no such instruction: `ldp q30,q31,[x0]'
sgemv_1x8_a72.S:83: Error: too many memory references for `and'
sgemv_1x8_a72.S:84: Error: no such instruction: `b.lt loop1'
sgemv_1x8_a72.S:85: Error: no such instruction: `lsr x6,x3,0x2'
sgemv_1x8_a72.S:90: Error: no such instruction: `ldr q0,[x1]'
sgemv_1x8_a72.S:91: Error: no such instruction: `ldp q16,q17,[x2]'
sgemv_1x8_a72.S:92: Error: no such instruction: `ldp q18,q19,[x2,0x20]'
sgemv_1x8_a72.S:93: Error: no such instruction: `prfm pldl1keep,[x1,0x80]'
sgemv_1x8_a72.S:94: Error: too many memory references for `add'
sgemv_1x8_a72.S:95: Error: too many memory references for `sub'
sgemv_1x8_a72.S:97: Error: no such instruction: `fmla v30.4s,v16.4s,v0.s[0]'
sgemv_1x8_a72.S:98: Error: no such instruction: `fmla v31.4s,v17.4s,v0.s[0]'
sgemv_1x8_a72.S:99: Error: no such instruction: `ldp q20,q21,[x2,0x40]'
sgemv_1x8_a72.S:100: Error: no such instruction: `fmla v30.4s,v18.4s,v0.s[1]'
sgemv_1x8_a72.S:101: Error: no such instruction: `prfm pldl1keep,[x2,0x400]'
sgemv_1x8_a72.S:102: Error: no such instruction: `fmla v31.4s,v19.4s,v0.s[1]'
sgemv_1x8_a72.S:103: Error: no such instruction: `ldp q22,q23,[x2,0x60]'
sgemv_1x8_a72.S:104: Error: no such instruction: `fmla v30.4s,v20.4s,v0.s[2]'
sgemv_1x8_a72.S:105: Error: no such instruction: `prfm pldl1keep,[x2,0x440]'
sgemv_1x8_a72.S:106: Error: too many memory references for `add'
sgemv_1x8_a72.S:107: Error: no such instruction: `fmla v31.4s,v21.4s,v0.s[2]'
sgemv_1x8_a72.S:108: Error: no such instruction: `fmla v30.4s,v22.4s,v0.s[3]'
sgemv_1x8_a72.S:109: Error: no such instruction: `fmla v31.4s,v23.4s,v0.s[3]'
sgemv_1x8_a72.S:111: Error: no such instruction: `b.ne loop4'
sgemv_1x8_a72.S:113: Error: no such instruction: `cbz x5,save_result'
sgemv_1x8_a72.S:116: Error: no such instruction: `ldr s0,[x1],0x4'
sgemv_1x8_a72.S:117: Error: no such instruction: `ldp q16,q17,[x2],0x20'
sgemv_1x8_a72.S:118: Error: too many memory references for `sub'
sgemv_1x8_a72.S:120: Error: no such instruction: `fmla v30.4s,v16.4s,v0.s[0]'
sgemv_1x8_a72.S:121: Error: no such instruction: `fmla v31.4s,v17.4s,v0.s[0]'
sgemv_1x8_a72.S:123: Error: no such instruction: `b.ne loop1'
sgemv_1x8_a72.S:126: Error: no such instruction: `stp q30,q31,[x4]'
Tengine/scripts/makefile.build:116: recipe for target '/Tengine/build/executor/operator/arm64/fc/sgemv_1x8_a72.o' failed
make[4]: *** [Tengine/build/executor/operator/arm64/fc/sgemv_1x8_a72.o] Error 1
make[4]: Leaving directory '/Tengine/executor/operator/arm64/fc'
/Tengine/scripts/makefile.build:72: recipe for target 'fc' failed
make[3]: *** [fc] Error 2
make[3]: Leaving directory '/Tengine/executor/operator/arm64'
/Tengine/scripts/makefile.build:72: recipe for target 'arm64' failed
make[2]: *** [arm64] Error 2
make[2]: Leaving directory '/Tengine/executor/operator'
/Tengine/scripts/makefile.build:72: recipe for target 'operator' failed
make[1]: *** [operator] Error 2
make[1]: Leaving directory '/Tengine/executor'
Makefile:166: recipe for target 'executor' failed
make: *** [executor] Error 2

clhne, 2018-09-07:
Thanks, Tengine Demo can run now.
But when build the example, result errors:
[image]

lyuchuny3, 2018-09-10:
If you want to run caffe_wrapper example, please verify your `makefile.config`:

- close CONFIG_CAFFE_REF: `# CONFIG_CAFFE_REF=y`
- no need to set CAFFE_ROOT: `# CAFFE_ROOT = /home/firefly/caffe`
- open CONFIG_CAFFE_SERIALIZER: `CONFIG_CAFFE_SERIALIZER=y`

Thanks.

closed 2018-09-25 by cyberfire