OAID/Tengine #47: build error . on RK3399, what's happend ...

bjthemost, 2018-09-22:
g++ -std=c++11    -I.  -Wno-ignored-attributes -Werror -O3 -Wall -g -fPIC  -I/media/firefly/uP/Tengine/driver/include  -I/media/firefly/uP/Tengine/driver/../include  -I/media/firefly/uP/Tengine/driver/../core/include -I/media/firefly/uP/Tengine/driver/../operator/include  -I/media/firefly/uP/Tengine/driver/../executor/include   -DCONFIG_ARCH_ARM64=1 -DCONFIG_ACL_GPU=1  -I/usr/local/AID/ComputeLibrary -I/usr/local/AID/ComputeLibrary/include -I/usr/local/AID/ComputeLibrary -c acl_graph_driver.cpp -o /media/firefly/uP/Tengine/build/driver/acl_graph/acl_graph_driver.o 
In file included from acl_graph_device.hpp:35:0,
                 from acl_graph_driver.hpp:36,
                 from acl_graph_driver.cpp:25:
acl_graph.hpp: In member function ‘bool TEngine::CLGraph::AddConvolutionLayer(TEngine::Node*)’:
acl_graph.hpp:360:23: error: no matching function for call to ‘arm_compute::ActivationLayerInfo::ActivationLayerInfo()’
   ActivationLayerInfo act_info; 
                       ^
In file included from /usr/local/AID/ComputeLibrary/arm_compute/core/ITensorInfo.h:30:0,
                 from acl_graph.hpp:34,
                 from acl_graph_device.hpp:35,
                 from acl_graph_driver.hpp:36,
                 from acl_graph_driver.cpp:25:
/usr/local/AID/ComputeLibrary/arm_compute/core/Types.h:676:5: note: candidate: arm_compute::ActivationLayerInfo::ActivationLayerInfo(arm_compute::ActivationLayerInfo::ActivationFunction, float, float)
     ActivationLayerInfo(ActivationFunction f, float a = 0.0f, float b = 0.0f)
     ^
/usr/local/AID/ComputeLibrary/arm_compute/core/Types.h:676:5: note:   candidate expects 3 arguments, 0 provided
/usr/local/AID/ComputeLibrary/arm_compute/core/Types.h:650:7: note: candidate: constexpr arm_compute::ActivationLayerInfo::ActivationLayerInfo(const arm_compute::ActivationLayerInfo&)
 class ActivationLayerInfo
       ^
/usr/local/AID/ComputeLibrary/arm_compute/core/Types.h:650:7: note:   candidate expects 1 argument, 0 provided
/usr/local/AID/ComputeLibrary/arm_compute/core/Types.h:650:7: note: candidate: constexpr arm_compute::ActivationLayerInfo::ActivationLayerInfo(arm_compute::ActivationLayerInfo&&)
/usr/local/AID/ComputeLibrary/arm_compute/core/Types.h:650:7: note:   candidate expects 1 argument, 0 provided
In file included from acl_graph_device.hpp:35:0,
                 from acl_graph_driver.hpp:36,
                 from acl_graph_driver.cpp:25:
acl_graph.hpp:429:65: error: no matching function for call to ‘arm_compute::CLDepthwiseConvolutionLayer3x3::configure(arm_compute::CLTensor*&, arm_compute::CLTensor*&, arm_compute::CLTensor*&, arm_compute::CLTensor*&, arm_compute::PadStrideInfo, int, arm_compute::ActivationLayerInfo&)’
          PadStrideInfo(stride_x,stride_y,pad_x,pad_y),1,act_info);
                                                                 ^
In file included from /usr/local/AID/ComputeLibrary/arm_compute/runtime/CL/CLFunctions.h:47:0,
                 from acl_graph.hpp:38,
                 from acl_graph_device.hpp:35,
                 from acl_graph_driver.hpp:36,
                 from acl_graph_driver.cpp:25:
/usr/local/AID/ComputeLibrary/arm_compute/runtime/CL/functions/CLDepthwiseConvolutionLayer.h:61:10: note: candidate: void arm_compute::CLDepthwiseConvolutionLayer3x3::configure(arm_compute::ICLTensor*, const arm_compute::ICLTensor*, const arm_compute::ICLTensor*, arm_compute::ICLTensor*, const arm_compute::PadStrideInfo&)
     void configure(ICLTensor *input, const ICLTensor *weights, const ICLTensor *biases, ICLTensor *output, const PadStrideInfo &conv_info);
          ^
/usr/local/AID/ComputeLibrary/arm_compute/runtime/CL/functions/CLDepthwiseConvolutionLayer.h:61:10: note:   candidate expects 5 arguments, 7 provided
In file included from acl_graph_device.hpp:35:0,
                 from acl_graph_driver.hpp:36,
                 from acl_graph_driver.cpp:25:
acl_graph.hpp:434:17: error: ‘class arm_compute::ActivationLayerInfo’ has no member named ‘enabled’
     if(act_info.enabled())
                 ^
acl_graph.hpp:447:43: error: no matching function for call to ‘arm_compute::CLConvolutionLayer::configure(arm_compute::CLTensor*&, arm_compute::CLTensor*&, arm_compute::CLTensor*&, arm_compute::CLTensor*&, arm_compute::PadStrideInfo, arm_compute::WeightsInfo, arm_compute::Size2D, arm_compute::ActivationLayerInfo&)’
      WeightsInfo(),Size2D(1U,1U), act_info);
                                           ^
In file included from /usr/local/AID/ComputeLibrary/arm_compute/runtime/CL/CLFunctions.h:44:0,
                 from acl_graph.hpp:38,
                 from acl_graph_device.hpp:35,
                 from acl_graph_driver.hpp:36,
                 from acl_graph_driver.cpp:25:
/usr/local/AID/ComputeLibrary/arm_compute/runtime/CL/functions/CLConvolutionLayer.h:108:10: note: candidate: void arm_compute::CLConvolutionLayer::configure(const arm_compute::ICLTensor*, const arm_compute::ICLTensor*, const arm_compute::ICLTensor*, arm_compute::ICLTensor*, const arm_compute::PadStrideInfo&, const arm_compute::WeightsInfo&)
     void configure(const ICLTensor *input, const ICLTensor *weights, const ICLTensor *biases, ICLTensor *output, const PadStrideInfo &conv_info, const WeightsInfo &weights_info = Weigh
          ^
/usr/local/AID/ComputeLibrary/arm_compute/runtime/CL/functions/CLConvolutionLayer.h:108:10: note:   candidate expects 6 arguments, 8 provided
In file included from acl_graph_device.hpp:35:0,
                 from acl_graph_driver.hpp:36,
                 from acl_graph_driver.cpp:25:
acl_graph.hpp: In member function ‘bool TEngine::CLGraph::AddPoolingLayer(TEngine::Node*)’:
acl_graph.hpp:663:93: error: no matching function for call to ‘arm_compute::PoolingLayerInfo::PoolingLayerInfo(arm_compute::PoolingType, arm_compute::Size2D, arm_compute::PadStrideInfo)’
                     PadStrideInfo(stride_x,stride_y,pad_x,pad_y,DimensionRoundingType::CEIL));
                                                                                             ^
In file included from /usr/local/AID/ComputeLibrary/arm_compute/core/ITensorInfo.h:30:0,
                 from acl_graph.hpp:34,
                 from acl_graph_device.hpp:35,
                 from acl_graph_driver.hpp:36,
                 from acl_graph_driver.cpp:25:
/usr/local/AID/ComputeLibrary/arm_compute/core/Types.h:583:14: note: candidate: arm_compute::PoolingLayerInfo::PoolingLayerInfo(arm_compute::PoolingType)
     explicit PoolingLayerInfo(PoolingType pool_type)
[... 26 lines cut ...]
         : _id{ { dims... } }, _num_dimensions{ sizeof...(dims) }
                                                                ^
/usr/local/AID/ComputeLibrary/arm_compute/core/Dimensions.h: In instantiation of ‘arm_compute::Dimensions<T>::Dimensions(Ts ...) [with Ts = {int, int, int, int}; T = long unsigned int]’:
/usr/local/AID/ComputeLibrary/arm_compute/core/TensorShape.h:47:31:   required from ‘arm_compute::TensorShape::TensorShape(Ts ...) [with Ts = {int, int, int, int}]’
acl_graph.hpp:171:75:   required from here
/usr/local/AID/ComputeLibrary/arm_compute/core/Dimensions.h:53:64: error: narrowing conversion of ‘dims#0’ from ‘int’ to ‘long unsigned int’ inside { } [-Werror=narrowing]
/usr/local/AID/ComputeLibrary/arm_compute/core/Dimensions.h:53:64: error: narrowing conversion of ‘dims#1’ from ‘int’ to ‘long unsigned int’ inside { } [-Werror=narrowing]
/usr/local/AID/ComputeLibrary/arm_compute/core/Dimensions.h:53:64: error: narrowing conversion of ‘dims#2’ from ‘int’ to ‘long unsigned int’ inside { } [-Werror=narrowing]
/usr/local/AID/ComputeLibrary/arm_compute/core/Dimensions.h:53:64: error: narrowing conversion of ‘dims#3’ from ‘int’ to ‘long unsigned int’ inside { } [-Werror=narrowing]
/usr/local/AID/ComputeLibrary/arm_compute/core/Dimensions.h: In instantiation of ‘arm_compute::Dimensions<T>::Dimensions(Ts ...) [with Ts = {int, int}; T = long unsigned int]’:
/usr/local/AID/ComputeLibrary/arm_compute/core/TensorShape.h:47:31:   required from ‘arm_compute::TensorShape::TensorShape(Ts ...) [with Ts = {int, int}]’
acl_graph.hpp:567:43:   required from here
/usr/local/AID/ComputeLibrary/arm_compute/core/Dimensions.h:53:64: error: narrowing conversion of ‘dims#0’ from ‘int’ to ‘long unsigned int’ inside { } [-Werror=narrowing]
/usr/local/AID/ComputeLibrary/arm_compute/core/Dimensions.h:53:64: error: narrowing conversion of ‘dims#1’ from ‘int’ to ‘long unsigned int’ inside { } [-Werror=narrowing]
At global scope:
[... cut ...]
/media/firefly/uP/Tengine/scripts/makefile.build:108: recipe for target '/media/firefly/uP/Tengine/build/driver/acl_graph/acl_graph_driver.o' failed
make[2]: *** [/media/firefly/uP/Tengine/build/driver/acl_graph/acl_graph_driver.o] Error 1
make[2]: Leaving directory '/media/firefly/uP/Tengine/driver/acl_graph'
/media/firefly/uP/Tengine/scripts/makefile.build:72: recipe for target 'acl_graph' failed
make[1]: *** [acl_graph] Error 2
make[1]: Leaving directory '/media/firefly/uP/Tengine/driver'
Makefile:166: recipe for target 'driver' failed
make: *** [driver] Error 2

cyberfire, 2018-09-25:
@bjthemost 

It is probably that you are using ACL 18.08.  We do not verify that version yet. Please try on ACL 18.05

Thanks,

Haitao

bjthemost, 2018-09-25:
Im 换成18.05了， 但是问题依旧， 最高版本 我们测到哪里？ 17.xx 是不是保险一些。
谢谢

bjthemost, 2018-09-26:
aclV18.05, V18.02 V17.12， same problem 。。。

hey-yahei, 2018-09-28:
**ACL v18.05** works here. 
Make sure that `ACL_ROOT` you set in `makefile.config` is the correct path of ACL v18.05.
If you already build Tengine, remember to remove the directory `build/driver` before rebuild.

bjthemost, 2018-09-29:
use below :
`````
cd ~/ComputeLibrary
git checkout 8a3da6f91f90c566b844d568f4ec43b946915af8
scons Werror=1 -j4 debug=0 asserts=1 neon=1 opencl=1 embed_kernels=1 os=linux arch=arm64-v8a
````````
is ok

root@firefly:/opt/Tengine1606/install/bin# ./bench_sqz 
ACL Driver Initialized
ACL Graph Initialized
Driver: ACLGraph probed 1 devices
Driver: ACLDriver probed 1 devices
tensor: prob created by node: prob is not consumed
add the node: prob into output list
Load model successfully
REPEAT COUNT= 100
Repeat [100] time 57888.04 us per RUN. used 5788804 us
0.2763 - "n02123045 tabby, tabby cat"
0.2673 - "n02123159 tiger cat"
0.1766 - "n02119789 kit fox, Vulpes macrotis"
0.0827 - "n02124075 Egyptian cat"
0.0777 - "n02085620 Chihuahua"
--------------------------------------
Release Graph Executor for graph graph0
ALL TEST DONE
Segmentation fault (core dumped)
root@firefly:/opt/Tengine1606/install/bin# ./bench_mobilenet 
ACL Driver Initialized
ACL Graph Initialized
Driver: ACLGraph probed 1 devices
Driver: ACLDriver probed 1 devices
tensor: fc7 created by node: fc7 is not consumed
add the node: fc7 into output list
Load model successfully
REPEAT COUNT= 100
Repeat [100] time 67250.52 us per RUN. used 6725052 us
8.5976 - "n02123159 tiger cat"
7.9550 - "n02119022 red fox, Vulpes vulpes"
7.8679 - "n02119789 kit fox, Vulpes macrotis"
7.4274 - "n02113023 Pembroke, Pembroke Welsh corgi"
6.3646 - "n02123045 tabby, tabby cat"
Release Graph Executor for graph graph0
ALL TEST DONE
Segmentation fault (core dumped)
root@firefly:/opt/Tengine1606/install/bin# ./bench_mobilenet 
ACL Driver Initialized
ACL Graph Initialized
Driver: ACLGraph probed 1 devices
Driver: ACLDriver probed 1 devices
tensor: fc7 created by node: fc7 is not consumed
add the node: fc7 into output list
Load model successfully
REPEAT COUNT= 100
Repeat [100] time 66645.90 us per RUN. used 6664590 us
8.5976 - "n02123159 tiger cat"
7.9550 - "n02119022 red fox, Vulpes vulpes"
7.8679 - "n02119789 kit fox, Vulpes macrotis"
7.4274 - "n02113023 Pembroke, Pembroke Welsh corgi"
6.3646 - "n02123045 tabby, tabby cat"
Release Graph Executor for graph graph0
ALL TEST DONE
Segmentation fault (core dumped)
root@firefly:/opt/Tengine1606/install/bin# cd /opt/ComputeLibrary/
root@firefly:/opt/ComputeLibrary# git status
HEAD detached at 8a3da6f
Untracked files:
  (use "git add <file>..." to include in what will be committed)
`````````
thank you everyone .