OAID/Tengine #48: Android Build

AbyShk95, 2018-09-24:
Hello!
I am trying to build Tengine for Android and have been stuck since long.
I was able to run and compile Tengine on Ubuntu.
I then moved on to modify the android_build_armv8.sh.
My question:

1. While adding protobuf path, I need to give path to bin, include or lib folder? (bin has protoc, include has google, lib has .a files )

2. Same question for BLAS path.

3. I trried various combinations of ARCH_TYPE. But I always get error. What architecture should I feed for different ABIs (specifically arm64-v8a) 

4. After modifying this file, should I do: i. cd build ii. ..android_build_armv8.sh

5. Is there an app which you provide? I was able to run Caffe-Mobile perfectly and their repo has an android app too.

Any help/ directions are appreciated. If there is a readme for android build, can that be shared too?

Thank you!

cyberfire, 2018-09-25:
@AbyShk95 

Please let me know if this doc https://github.com/OAID/Tengine/blob/master/doc/build_android.md could help.

Thanks,

Haitao

WenguoLi, 2018-10-11:
I encountered the same problem, build error according to https://github.com/OAID/Tengine/blob/master/doc/build_android.md 

(1)  CMake , 
apuser@apuser-H81-M1:~/deeplearning/tengine/build_armv8$ ../android_build_armv8.sh
-- CMAKE_TOOLCHAIN_FILE = /home/apuser/android-ndk/android-ndk-r16b/build/cmake/android.toolchain.cmake
-- CMAKE_INSTALL_PREFIX = /home/apuser/deeplearning/tengine/install
-- Check for working C compiler: /home/apuser/android-ndk/android-ndk-r16b/toolchains/llvm/prebuilt/linux-x86_64/bin/clang
-- Check for working C compiler: /home/apuser/android-ndk/android-ndk-r16b/toolchains/llvm/prebuilt/linux-x86_64/bin/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/apuser/android-ndk/android-ndk-r16b/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++
-- Check for working CXX compiler: /home/apuser/android-ndk/android-ndk-r16b/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
GIT COMMIT ID: 0xfbf368ff71bb8883ded2b7bc6de4da1a18145ef7
CAFFE SERIALIZER DEFINED: /home/apuser/deeplearning/tengine/build_armv8/serializer/caffe/te_caffe.pb.cc
-- The ASM-ATT compiler identification is GNU
-- Found assembler: /home/apuser/android-ndk/android-ndk-r16b/toolchains/aarch64-linux-android-4.9/prebuilt/linux-x86_64/bin/aarch64-linux-android-as
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    BLAS_DIR

-- Build files have been written to: /home/apuser/deeplearning/tengine/build_armv8

(2)  make 

[ 40%] Building CXX object CMakeFiles/tengine.dir/operator/operator/split.cpp.o
[ 41%] Building CXX object CMakeFiles/tengine.dir/operator/plugin/init.cpp.o
[ 42%] Building CXX object CMakeFiles/tengine.dir/serializer/caffe/caffe_serializer.cpp.o
In file included from /home/apuser/deeplearning/tengine/serializer/caffe/caffe_serializer.cpp:37:
In file included from /home/apuser/deeplearning/tengine/serializer/include/caffe_serializer.hpp:32:
/home/apuser/deeplearning/tengine/serializer/include/te_caffe.pb.h:17:2: error: 
      This file was generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
/home/apuser/deeplearning/tengine/serializer/include/te_caffe.pb.h:18:2: error: 
      incompatible with your Protocol Buffer headers. Please
#error incompatible with your Protocol Buffer headers.  Please
 ^
/home/apuser/deeplearning/tengine/serializer/include/te_caffe.pb.h:19:2: error: 
      regenerate this file with a newer version of protoc.
#error regenerate this file with a newer version of protoc.
 ^
3 errors generated.
CMakeFiles/tengine.dir/build.make:801: recipe for target 'CMakeFiles/tengine.dir/serializer/caffe/caffe_serializer.cpp.o' failed
make[2]: *** [CMakeFiles/tengine.dir/serializer/caffe/caffe_serializer.cpp.o] Error 1
CMakeFiles/Makefile2:137: recipe for target 'CMakeFiles/tengine.dir/all' failed
make[1]: *** [CMakeFiles/tengine.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

(3) protobuf version
 protoc --version
libprotoc 2.6.1

cyberfire, 2018-10-15:
@WenguoLi 

The pre-built libprotobuf library is 3.0.0. so, please make sure protoc in your host is 3.0.0 too.

Thanks,

Haitao

WenguoLi, 2018-10-16:
Thanks @cyberfire