OAID/Tengine #58: Tengine build issue with ACL.

rahulsharma11, 2018-10-26:
Hi,
While enabling the ACL path in makefile.config in Tengine, it shows error while make-
In file included from acl_graph_device.hpp:35:0,
                 from acl_graph_driver.hpp:36,
                 from acl_graph_driver.cpp:25:
acl_graph.hpp: In member function ‘bool TEngine::CLGraph::AddFCLayer(TEngine::Node*)’:
acl_graph.hpp:595:69: error: no matching function for call to ‘arm_compute::CLFullyConnectedLayer::configure(arm_compute::CLTensor*&, arm_compute::CLTensor*&, arm_compute::CLTensor*&, arm_compute::CLTensor*&, bool&)’
         fc->configure(itensor, wtensor, btensor, otensor,transpose_w);

Tested with ACL ver-18.05 and 18.08.
Please suggest any missing step.
Thanks.

rahulsharma11, 2018-10-26:
Update-
Build is successfull.
Please let me know how to include ACL with any app.
such as i ma running ssd model, so should it be like "./SSD acl_opencl"?

As running ./build/tests/bin/bench_sqz -r1 gives error-
Register reference's operators
terminate called after throwing an instance of 'cl::Error'
  what():  empty
Aborted

I am using Ubuntu18 with ACL version18.05.
Thanks.

hey-yahei, 2018-10-26:
Reference: [document for ACL Driver](https://github.com/OAID/Tengine/blob/master/doc/acl_driver.md)
1. Firstly, can you successfully run `./build/tests/bin/bench_sqz` with parameter `-d acl_opencl` to set gpu as default device?
2. To include ACL with any app, call the function `set_default_device("acl_opencl")` in your own app just as [tests/bin/bench_sqz.cpp](https://github.com/OAID/Tengine/blob/master/tests/bin/bench_sqz.cpp#L202) do.
3. If running several graphs in one app is needed, replace `set_default_device` function with `set_graph_device` to set different device for different graph.

rahulsharma11, 2018-10-26:
Hi,
Thanks for the reply.
I don't know what is happening, i am using rockpro64 board (RK3399). I did checked out ACL vers 18.05 and built that. But now not able to build Tengine again. As according to https://github.com/OAID/Tengine/blob/master/doc/acl_driver.md 
i remove build/driver and then build again. Now it is throwing error every time-
In file included from acl_graph_device.hpp:35:0,
                 from acl_graph_driver.hpp:36,
                 from acl_graph_driver.cpp:25:
acl_graph.hpp: In member function ‘bool TEngine::CLGraph::AddConvolutionLayer(TEngine::Node*)’:
acl_graph.hpp:360:23: error: no matching function for call to ‘arm_compute::ActivationLayerInfo::ActivationLayerInfo()’
   ActivationLayerInfo act_info;
....

2)Second doubt is-
In Tengine/example/CMakeList, there is a line-
if( NOT ANDROID)
   add_subdirectory(caffe_wrapper)
endif()

If i comment these lines then it successfully build (WITHOUT ACL), and if i uncomment these lines it through error of "cublas_v2.h" : no such file or directory.

hey-yahei, 2018-10-27:
Hi, rahulsharma11

1. Just double check, make sure that your ACL version is v18.05? As I know, class `arm_compute::ActivationLayerInfo` is added default constructor with no parameters after v18.02. 
2. If the version of ACL is right, try to reclone Tengine from github to your devices, check your path to ACL in `makefile.config` and `make clean && make -j4` again. 
BTW, why not show your settings in `makefile.config` file which may help debug?
3. Confused, `cublas.h` seems a header for CUDA library, and I can't find any source code of Tengine with `cublas.h` included. Can you show me more error information? For example, which source file includes it?

Thanks

rahulsharma11, 2018-10-29:
Hi,
i did checkout v18.05 for ACL, and did make clean and make again for Tengine.
It is successfully built.
I also did Sudo make install in Tengine to build examples.
Now while building examples, i am getting the error for cublas error.
1) Firstly i built caffe separately from-
https://github.com/BVLC/caffe
And did - 
mkdir build
cd build
cmake ..
make all
make install
make runtest
Then while building examples, it was throwing error for caffe/caffe.h not found, so included the caffe  path in CMakeList file in "Tengine/examples/caffe_wrapper/cpp_classification" and "Tengine/examples/caffe_wrapper/mtcnn". But now the error is for cublas again-
here is the output-
In file included from /home/rock64/rahul/caffe/include/caffe/common.hpp:19:0,
                 from /home/rock64/rahul/caffe/include/caffe/blob.hpp:8,
                 from /home/rock64/rahul/caffe/include/caffe/caffe.hpp:7,
                 from /home/rock64/rahul/Tengine/examples/caffe_wrapper/mtcnn/caffe_mtcnn.hpp:27,
                 from /home/rock64/rahul/Tengine/examples/caffe_wrapper/mtcnn/test_caffe_mtcnn.cpp:24:
/home/rock64/rahul/caffe/include/caffe/util/device_alternate.hpp:34:10: fatal error: cublas_v2.h: No such file or directory
 #include <cublas_v2.h>
          ^~~~~~~~~~~~~
compilation terminated.
caffe_wrapper/mtcnn/CMakeFiles/CAFFE_MTCNN.dir/build.make:62: recipe for target 'caffe_wrapper/mtcnn/CMakeFiles/CAFFE_MTCNN.dir/test_caffe_mtcnn.cpp.o' failed
make[2]: *** [caffe_wrapper/mtcnn/CMakeFiles/CAFFE_MTCNN.dir/test_caffe_mtcnn.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/rock64/rahul/caffe/include/caffe/common.hpp:19:0,
                 from /home/rock64/rahul/caffe/include/caffe/blob.hpp:8,
                 from /home/rock64/rahul/caffe/include/caffe/caffe.hpp:7,
                 from /home/rock64/rahul/Tengine/examples/caffe_wrapper/mtcnn/caffe_mtcnn.hpp:27,
                 from /home/rock64/rahul/Tengine/examples/caffe_wrapper/mtcnn/caffe_mtcnn.cpp:24:
/home/rock64/rahul/caffe/include/caffe/util/device_alternate.hpp:34:10: fatal error: cublas_v2.h: No such file or directory
 #include <cublas_v2.h>

rahulsharma11, 2018-10-29:
Changes in makefile.config file for Tengine-

# Set the path of caffe used for compile and link,
# if caffe's operator calculation is enabled.
 CAFFE_ROOT = /home/rock64/rahul/caffe

# Enable GPU support by Arm Computing Library
 CONFIG_ACL_GPU=y
#

# Use BLAS as the operator implementation
#
 CONFIG_ARCH_BLAS=y

# Set the path of ACL 
 ACL_ROOT=/home/rock64/rahul/ComputeLibrary

hey-yahei, 2018-10-29:
Hi, rahulsharma11

Maybe you did not uncomment `CONFIG_CAFFE_REF=y`.
`CONFIG_CAFFE_REF` and `CAFFE_ROOT` should be setted at the same time.

rahulsharma11, 2018-10-29:
Oh sry, i forget to mention that also.
Yes i did uncomment CONFIG_CAFFE_REF=y.

rahulsharma11, 2018-10-29:
And again i got error for

./build/tests/bin/bench_sqz -r1 gives error-
Register reference's operators
terminate called after throwing an instance of 'cl::Error'
what(): empty
Aborted

Now i am building Tengine again without ACL to double check the result.

hey-yahei, 2018-10-29:
Try to comment `CONFIG_CAFFE_REF=y` and `CAFFE_ROOT` to not build the reference operators wrapped on Caffe, then rebuild Tengine and examples again.

rahulsharma11, 2018-10-29:
Ok.
Will try that.

Also i confirm that if i build Tengine without ACL, then on ruuning "./build/tests/bin/bench_sqz -r1"
it gives successful results-
Register reference's operators
Register Driver: CaffeNode
Caffe Node Driver Initialized
Driver: CaffeNode probed 1 devices
tensor: prob created by node: prob is not consumed
add the node: prob into output list
Load model successfully
REPEAT COUNT= 1
Repeat [1] time 51573.00 us per RUN. used 51573 us
0.2763 - "n02123045 tabby, tabby cat"
0.2673 - "n02123159 tiger cat"
0.1766 - "n02119789 kit fox, Vulpes macrotis"
0.0827 - "n02124075 Egyptian cat"
0.0777 - "n02085620 Chihuahua"
--------------------------------------
Release Graph Executor for graph graph0
ALL TEST DONE
Release workspace default resource

And Examples are also build successfully.

rahulsharma11, 2018-10-29:
Hi,
I rebuilt Tengine by commneting Caffe and uncommnet ACL.
So built is successfull but  
./build/tests/bin/bench_sqz -r1 throws-
terminate called after throwing an instance of 'cl::Error'
  what():  empty
Aborted

cyberfire, 2018-10-30:
@rahulsharma11 

It looks that it is probably that you forget to "make install" after building tengine library.
Would you please check that?

Thanks,

Haitao

rahulsharma11, 2018-10-30:
Hi,
Yes i did "make install" after tengine make. 
I have freshly cloned both Compute library and Tengine. Builded Computelibrary by-
"aarch64-linux-gnu-gcc opencl-1.2-stubs/opencl_stubs.c -Iinclude -shared -o build/libOpenCL.so
scons Werror=0 -j4 debug=0 asserts=1 neon=1 opencl=1 embed_kernels=1 os=linux arch=arm64-v8a
wget ftp://ftp.openailab.net/tools/script/Computelibrary/Makefile
sudo make install"

Then in Tengine makefile.config, provided the ACL path. Tengine is built successfully.
Issue is with Tengine/Eaxmples.
Can you please confirm this for examples?
I am getting same error-
terminate called after throwing an instance of 'cl::Error'
what(): empty
Aborted

Without ACL everything is fine.
Thanks,

rahulsharma11, 2018-11-02:
Hi,
Now it is working.
What i did-
1) Rebuild Compute library with "neon=0".
2) Downloaded libmali from "https://goo.gl/25dk2r"
3) linked /usr/lib/aarch64-linux-gnu/libOpenCL.so.1.0.0 with "libmali-midgaurd-t86-r14p0-fbdev.so" in lib folder.
Then it worked.
Thanks for the support.