OAID/Tengine #62: ARM32 - RPI3B+ runtime error --> undefined symbol: _ZN7TEngine18RegisterConv2dFastEv

ghimiredhikura, 2018-11-06:
Device: RaspberryPi3B+ 
Raspbian, ARM32, armv7-a

In config file I set `CONFIG_ARCH_ARM32=y`
There was no error in library compilation/build. 
But at run time I got following error. 
```ruby
./build/tests/bin/bench_mobilenet
```
Output
```ruby
./build/tests/bin/bench_mobilenet: symbol lookup error: ./build/libtengine.so: undefined symbol: _ZN7TEngine18RegisterConv2dFastEv
```

Any help please? 
Thanks

Deepak

lyuchuny3, 2018-11-07:
To run Tengine on armv7, the makefile.config should:
* close `# CONFIG_ARCH_ARM64=y`
* open `CONFIG_ARCH_BLAS=y`

Thanks

ghimiredhikura, 2018-11-07:
Hello @lyuchuny3,

Thank you for reply. Yes, I already used Tengine with `CONFIG_ARCH_BLAS=y` successfully. 
But it looks like it is not optimized in `CONFIG_ARCH_BLAS=y` mode. 

It took 3462.49ms to run mobilenetv1. 

Any suggestion? 

Best,
Deepak

lyuchuny3, 2018-11-08:
How many cpus in RaspberryPi3B+? If four, try:
> export TENGINE_CPU_LIST=0,1,2,3

to enable tengine run on 4 cores.

ghimiredhikura, 2018-11-09:
I used single cpu. But still 3462.49ms for mobilenetv1 using single cpu isn't slow?
Yes, if all cpu are enabled it runs faster. 
 
Btw, I did rebuild openblas from source (https://github.com/xianyi/OpenBLAS) and build Tengine with new openblas, this time inference time for mobilenetv1 is 2295.61ms. But still not satisfactory!

closed 2019-01-18 by cyberfire