OAID/Tengine #64: ./SSD -d acl_opencl issue

rahulsharma11, 2018-11-12:


rahulsharma11, 2018-11-12:
Hi,
I am running Tengine on my rockpro64. All good while build. Also have built Tengine examples with ACL.
But whenever i run-
./SSD -d acl_opencl, 
it throws error - 
"load model done!
terminate called after throwing an instance of 'std::runtime_error'
  what():  in scaled_dimensions src/core/Utils.cpp:342: ((w - 1) * stride_x) >= (width + pad_left)
Aborted".

I tested the Compute library example separatly. ACL version-v18.05.
My makefile.config-

CONFIG_ARCH_ARM64=y
#Enable Compiling Optimization
CONFIG_OPT_CFLAGS = -O2

#uncomment to build the reference operators wrapped on Caffe
#CONFIG_CAFFE_REF=y

#Set the path of caffe used for compile and link,
#if caffe's operator calculation is enabled.
#CAFFE_ROOT = /home/rock64/rahul/caffe

#Enable GPU support by Arm Computing Library
CONFIG_ACL_GPU=y
#Use BLAS as the operator implementation
#CONFIG_ARCH_BLAS=y

#Set the path of ACL 
ACL_ROOT=/home/rock64/rahul/ComputeLibrary

#Enable other serializers 
CONFIG_CAFFE_SERIALIZER=y
#CONFIG_MXNET_SERIALIZER=y
#CONFIG_ONNX_SERIALIZER=y
#CONFIG_TF_SERIALIZER=y
CONFIG_TENGINE_SERIALIZER=y

It is running good without acl_opencl parameter.
What could be the reason? any suggestion.
Thanks.

rahulsharma11, 2018-11-13:
Hi,
Thanks for the clarification.
1) I followed this link to use ACL library- "https://github.com/OAID/Tengine/blob/master/doc/acl_driver.md", where i need to provide extra device parameter as argument. I did that hence i was thinking it should work. 
Although if SSD doesn't suuport ACL properly then its fine not to go with this further.

I was wondering to use rockpro64 at its full effeciency as on running SSD model for object detection, its running only with half of the load.

lyuchuny3, 2018-11-16:
Hi, rahulsharma11,
The lastest commit of Tengine has	support GPU/CPU Heterogeneous Computing, you can run MobilenetSSD as the doc described https://github.com/OAID/Tengine/blob/master/doc/gpu_cpu_mssd.md

Thanks.

rahulsharma11, 2018-11-16:
Hi, i tested this and its giving time consomption of around 250ms per run. There i can see cpu load is very less.
Thanks

closed 2018-11-16 by rahulsharma11