OAID/Tengine #66: How to develop an Android app with Tengine

vincent8877, 2018-11-15:
Hi
I already run my mobilenet_ssd model on firefly-RK3399 (ubuntu platform) with the help of hey-yahei
http://hey-yahei.cn/2018/08/04/RK3399-Tengine/
and it works well
now I want to run the object detection model on Android platform, and I look into the Android_build page:
https://github.com/OAID/Tengine/blob/master/doc/build_android.md
Seems that it can only run a classifer model using adb.
But I want develop an app with Android Studio, can anybody help me to get that?

cyberfire, 2018-11-16:
@vincent8877 

We will provide an Android APP soon.

Thanks,

Haitao

vincent8877, 2018-11-17:
expect

hey-yahei, 2018-12-18:
Hi, vincent8877
Tengine demo for Android Studio is provided at [Tengine-app](https://github.com/OAID/Tengine-app/tree/master/android/classification) now.

vincent8877, 2018-12-18:
great，thanks

closed 2018-12-31 by cyberfire