OAID/Tengine #67: Increasing clock governor frequencies.

rahulsharma11, 2018-11-19:
Hi,
I followed this link-
https://github.com/OAID/Tengine/blob/master/doc/gpu_cpu_mssd.md
to set the GPU frequency. Please clarify some of my doubts-
1) Similar to nvidia jetson "clock.sh" script which is used to maximize the performance of the device is there any script for rockpro64 also?
2) Can i set CPU frreq also or it is default 1200Mhz?
3) How can i run a project (say mobilenet_ssd with ACL) by consumng both CPU and GPU to increase its performance? Can i monitor that also?
Thanks.

lyuchuny3, 2018-11-26:
Hi, 
I try to answer you questions, if still have any problems, you can reply below:
1.  you write below scripts into a `clock.sh` and run it with sudo:
```
echo "performance" >/sys/devices/platform/ff9a0000.gpu/devfreq/ff9a0000.gpu/governor
cat /sys/devices/platform/ff9a0000.gpu/devfreq/ff9a0000.gpu/cur_freq
```
2. you can set cpu frequency
* for A72 (big core CPU)
```
echo performance > /sys/devices/system/cpu/cpufreq/policy4/scaling_governor 
cat /sys/devices/system/cpu/cpufreq/policy4/cpuinfo_cur_freq  
```
* for A53 (small core CPU)
```
echo performance > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor   
cat /sys/devices/system/cpu/cpufreq/policy0/cpuinfo_cur_freq
```
3.  you can run a project (mobilenet-ssd) with both cpu and gpu as this doc 
https://github.com/OAID/Tengine/wiki/GPU-CPU%E5%BC%82%E6%9E%84%E8%B0%83%E5%BA%A6%E8%B7%91MobilenetSSD

you can monitor by the **time used** to run this network:
* only use 1A53 CPU : 
```
export TENGINE_CPU_LIST=0
taskset 0x1 ./MSSD
```
* use both 1A53CPU and GPU
```
export GPU_CONCAT=0           # disable gpu run concat,     avoid frequent data transfer between cpu and gpu
export ACL_FP16=1             # enable gpu fp16
export REPEAT_COUNT=50        # repeat count to run mssd,     get avg time
taskset 0x1 ./MSSD -d acl_opencl   
``` 
it will output `repeat 50 times, avg time per run is xxx ms`

rahulsharma11, 2018-11-27:
Hi,
Thanks for the valuable reply.
I did this and checked. 
What i wanted is to monitor the GPU usage similar to what htop shows cpu usage.
I am sure that GPU is being utilised as "ACL graph Initialisation" is being printed on Terminal. But how can i visualize that?

closed 2019-01-15 by cyberfire