OAID/Tengine #69: How to integrate FaceDetection with Tengine engine.

rahulsharma11, 2018-11-22:
Hi,
I have used both Tengine and FaceDetection separately. FaceDetection is working good with caffe framework but not with mxnet-hrt. As i feel mxnet-hrt is a bit old thus is not being well supported. So i wanted to run Facedetection as a part inside Tengine like ssd, mobile_ssd. 
Any suggestion how can i integrate this?
Thanks.

rahulsharma11, 2018-11-22:
Hi, Thanks for the suggestion.
I want to integrate this as a makefile part of Tengine. So thus i need to do changes in CMakeList.txt also.
Can you help me out with that?

rahulsharma11, 2018-11-22:
Ok i got it.
I can check test_mtcnn.cpp for that. This will work similar to what facedetection project is doing. Is it so?
One more doubt-
In facedetection project, i have to pass some DL framework like caffe/mxnet. Here inside Tengine is it necessory?

hey-yahei, 2018-11-22:
It's not necessary to pass any DL framework since Tengine will parse the model of DL framework such as caffe and mxnet and then run it.

rahulsharma11, 2018-11-23:
Hi, 
Thanks for the clarification.
Its woring at around 20 fps.
But similar to FaceDetection i coudn't find FaceRecognition example file inside Tengine.

rahulsharma11, 2018-11-23:
Hi, While checking mtcnn.cpp file, it uses caffemodel and corresponding prototxt file (such as det1.caffemodel/det1/prototxt). How can i use mxnet model? You told me not necessary to pass DL framework but it seems it is using caffemodel bydefault. 
Also how can i be sure that it is using ACL or not.
Thanks

rahulsharma11, 2018-11-26:
Hi,
Thanks.

closed 2018-11-26 by rahulsharma11