OAID/Tengine #71: Gender-age detection inside Tengine.

rahulsharma11, 2018-11-23:
Hi,
Thanks for the support till now.
Face detection is running good inside Tengine.
For Gender- age detection, i was checking this link -"https://github.com/deepinsight/insightface/tree/master/gender-age".
That uses very light mxnet based model for gender-age detection. 
So my question is something similar to it already exist inside Tengine?
Thanks.

hey-yahei, 2018-11-24:
Hi, rahulsharma11
There are no examples of gender-age detection yet.
However, if operators of your model are all supported, Tengine would load and run it successfully.
Operators that Tengine support are listed [here](https://github.com/OAID/Tengine/blob/master/doc/operator_ir.md).
Thanks

rahulsharma11, 2018-11-26:
Hi, 
Thanks for reply.

Can i use Caffe-HRT with Tengine instead of caffe? As i have built Caffe-HRT for FaceDetection and Recognition. Now when i am building Tengine with Caffe support, i did in makefile.config of Tengine-
 # uncomment to build the reference operators wrapped on Caffe
   CONFIG_CAFFE_REF=y
  
 # Set the path of caffe used for compile and link,
 # if caffe's operator calculation is enabled.
 CAFFE_ROOT = /home/rock64/rahul/Caffe-HRT

Tengine is built successfully but when i make examples , it throws error-
 20%] Linking CXX executable FASTER_RCNN
[ 22%] Linking CXX executable YOLOV2
[ 25%] Building CXX object ssd/CMakeFiles/SSD.dir/__/common/common.cpp.o
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::CLConvolutionLayer::configure(arm_compute::ICLTensor const*, arm_compute::ICLTensor const*, arm_compute::ICLTensor const*, arm_compute::ICLTensor*, arm_compute::PadStrideInfo const&, arm_compute::WeightsInfo const&)'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::NEPoolingLayer::NEPoolingLayer()'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::CLDirectConvolutionLayer::configure(arm_compute::ICLTensor*, arm_compute::ICLTensor const*, arm_compute::ICLTensor const*, arm_compute::ICLTensor*, arm_compute::PadStrideInfo const&)'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `typeinfo for arm_compute::INESimpleFunction'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::NEBatchNormalizationLayer::NEBatchNormalizationLayer()'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::NEActivationLayer::configure(arm_compute::ITensor*, arm_compute::ITensor*, arm_compute::ActivationLayerInfo)'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::NEDirectConvolutionLayer::configure(arm_compute::ITensor*, arm_compute::ITensor const*, arm_compute::ITensor const*, arm_compute::ITensor*, arm_compute::PadStrideInfo const&)'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::INESimpleFunction::INESimpleFunction()'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::NEFullyConnectedLayer::configure(arm_compute::ITensor const*, arm_compute::ITensor const*, arm_compute::ITensor const*, arm_compute::ITensor*, bool, bool)'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::NEDepthConcatenateLayer::NEDepthConcatenateLayer()'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::CLBatchNormalizationLayer::configure(arm_compute::ICLTensor*, arm_compute::ICLTensor*, arm_compute::ICLTensor const*, arm_compute::ICLTensor const*, arm_compute::ICLTensor const*, arm_compute::ICLTensor const*, float)'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `vtable for arm_compute::INESimpleFunction'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::NEDirectConvolutionLayer::NEDirectConvolutionLayer(std::shared_ptr<arm_compute::IMemoryManager>)'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::NELocallyConnectedLayer::NELocallyConnectedLayer(std::shared_ptr<arm_compute::IMemoryManager>)'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::NENormalizationLayer::configure(arm_compute::ITensor const*, arm_compute::ITensor*, arm_compute::NormalizationLayerInfo const&)'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::NELocallyConnectedLayer::configure(arm_compute::ITensor const*, arm_compute::ITensor const*, arm_compute::ITensor const*, arm_compute::ITensor*, arm_compute::PadStrideInfo const&)'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::INESimpleFunction::run()'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::CLSubTensor::CLSubTensor(arm_compute::ICLTensor*, arm_compute::TensorShape const&, arm_compute::Coordinates const&)'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::NENormalizationLayer::NENormalizationLayer(std::shared_ptr<arm_compute::IMemoryManager>)'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::NEConvolutionLayer::NEConvolutionLayer(std::shared_ptr<arm_compute::IMemoryManager>)'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::NEBatchNormalizationLayer::configure(arm_compute::ITensor*, arm_compute::ITensor*, arm_compute::ITensor const*, arm_compute::ITensor const*, arm_compute::ITensor const*, arm_compute::ITensor const*, float)'
//home/rock64/rahul/Caffe-HRT/build/lib/libcaffe.so.1.0.0-rc5: undefined reference to `arm_compute::NEDepthConcatenateLayer::configure(std::vector<arm_compute::ITensor*, std::allocator<arm_compute::ITensor*> >, arm_compute::ITensor*)'
[... cut ...]
make[2]: *** [mtcnn/CMakeFiles/MTCNN.dir/test_mtcnn.cpp.o] Interrupt
make[2]: *** [ssd/CMakeFiles/SSD.dir/__/multiwebcam/multiWebCam.cpp.o] Interrupt
CMakeFiles/Makefile2:195: recipe for target 'ssd/CMakeFiles/SSD.dir/all' failed
make[1]: *** [ssd/CMakeFiles/SSD.dir/all] Interrupt
CMakeFiles/Makefile2:85: recipe for target 'mtcnn/CMakeFiles/MTCNN.dir/all' failed
make[1]: *** [mtcnn/CMakeFiles/MTCNN.dir/all] Interrupt
Makefile:83: recipe for target 'all' failed
make: *** [all] Interrupt

Is there any version issue? 
Thanks.

lyuchuny3, 2018-11-26:
Hi， 
I have update tengine doc, rm `CAFFE_REF` in  `makefile.config.example`,
you can pull the lastest Tengine，and rebuild your project.

Caffe model support is by the key `CONFIG_CAFFE_SERIALIZER=y` (which default is open)   https://github.com/OAID/Tengine/blob/master/makefile.config.example#L37

thanks

rahulsharma11, 2018-11-27:
Hi, 
I want to continue this thread with one last question.
I am integrating classification example of Tengine caffe_wrapper with test_mtcnn.cpp.
I have done the following-
1) added include path of wrapper_include in CMakeList of mtcnn. (for caffe.hpp header)
2) copied some part of classification.cpp code such as classifier, preprocess and all necessary functions for classification in test_mtcnn.cpp. 
Is it correct to do? as while make, It is throwing error for  undefined references like-
In function 'Classifier::Preprocess(cv::Mat const& std::vector<cv::Mat std::allocator<cv::Mat> >*)':
/home/rahul/Tengine/examples/mtcnn/test_mtcnn.cpp:258: undefined reference to 'caffe::Blob_wrap<float>::cpu_data() const'
/home/rahul/Tengine/examples/mtcnn/test_mtcnn.cpp:258: undefined reference to 'google::LogMessageFatal::LogMessageFatal(char const*, int)' .......
and so on....
Any suggestion?
Thanks.

lyuchuny3, 2018-12-07:
Hi，
If you want to run https://github.com/OAID/Tengine/tree/master/examples/caffe_wrapper/mtcnn

## Step1, install tengine with wrapper
If your want to run tengine caffe_wrapper, in your `makefile.config`, you should open:
```
# Enable Wrappers
CONFIG_FRAMEWORK_WRAPPER=y
``` 
and then  `make & make install`, this will auto copy  `wrapper_include ` into `include/` directory
##  Step2. build caffe_wrapper example
```
cd caffe_wrapper/mtcnn
mkdir build
cd build
cmake -DTENGINE_DIR=/home/your_dir/tengine ..
make
``` 

If you still have problem, you can reply below.
Thanks.