OAID/Tengine #73: How can I train custom mobilenet ssd model?

gunjupark, 2018-11-27:
I want to get custom MobileNetSSD_deploy.caffemodel which can be run on Tengine

So I trained my custom model following this referrence site 
https://github.com/chuanqi305/MobileNet-SSD

But in Tengine's MobileNet prototxt , BatchNorm op doesn't exist.

so could you upload your MobileNet_train.prototxt 

Thank you

gunjupark, 2018-11-27:
I found no_bn.prototxt 
thankyou

lyuchuny3, 2018-11-27:
Hi， 
You can run `merge_bn` to merge batchnorm op into conv
```
python merge_bn.py --model example/MobileNetSSD_deploy.prototxt --weights snapshot/mobilenet_iter_xxxxxx.caffemodel
```
Details see https://github.com/chuanqi305/MobileNet-SSD/blob/master/README.md#train-your-own-dataset   step6.

closed 2018-12-03 by gunjupark

KrishnaShalini0518-arch, 2021-04-08:
Hi @gunjupark,

Could you share the references you used for training Tengine model?