OAID/Tengine #74: caffe/caffe.hpp : no such file or directory in Tengine example build.

rahulsharma11, 2018-11-27:
Hi,
While tengine/examples build, it throws error for -
examples/caffe_wrapper/cpp_classification/classification.cpp:1:10: fatal error: caffe/caffe.hpp: No such file or directory
#include<caffe/caffe.hpp>

I have enabled the CONFIG_CAFFE_SERILIZER option in makefile.config in Tengine main directory. 
Also in Latest Tengine, there is no need to provide the caffe_path separately. So i am assuming there must be internal caffe libraries in Tengine.

It builds successfully when i comment the following in examples/CMakeList.txt file -
if(NOT ANDROID)
     add_subdirectory(caffe_wrapper)
endif()

Any path i need to provide for  caffe_wrapper?
Thanks.

rahulsharma11, 2018-11-27:
Hi,
Update-
There was mistake in CMakeList.txt file in cpp_classification folder inside Tengine-
original - "include_directories(${INSTALL_DIR}/wrapper_include"
changed - "include_directories(${INSTALL_DIR}/../wrapper/include"

Then it built successfully.
But now a new issue-

lyuchuny3, 2018-11-27:
Hi, 
* `CONFIG_CAFFE_SERILIZER`  is use for serializer (parser) caffemodel
* to enable caffe_wrapper, you should open
`CONFIG_FRAMEWORK_WRAPPER=y`
then `make & make install` will auto copy needed headers into directory `install/`

rahulsharma11, 2018-11-27:
Ohhhhh.....
Thank you very much..:-)

closed 2018-11-27 by rahulsharma11