OAID/Tengine #75: Run tensorflow model on Android platfrom

Lemiron24, 2018-11-28:
Hi, 
I have run Tengine success on Andoird phone: imagenet_classification test with caffe model.
command: ./Classify -n inception_v3 -g 299,299 -i cat.jpg -r 10
Now I want to run the test with tensorflow model(.pb)
I open the macro: CONFIG_TF_SERIALIZER to make Tengine support tensorflow serializer.

besides all the models download from: Tengine model zoo

Could you give some help.

thanks

hey-yahei, 2018-11-29:
Hi, Lemiron24
You should modify parameters of **load_model** function in [classification.cpp#L163](https://github.com/OAID/Tengine/blob/master/examples/imagenet_classification/classification.cpp#L163) to specify the type of model as tensorflow, just as [test_tf_mobilenet.cpp#L116](https://github.com/OAID/Tengine/blob/master/tests/bin/test_tf_mobilenet.cpp#L116) do.
And rebuild examples.

Lemiron24, 2018-11-30:
@hey-yahei 
Hi, hey-yahei,

Thanks for your reply.
Now I am writing a apk which native c++ layer using Tengine including: parse the network model file in Tengine, kernel implement in Tengine. because I am not familiar with JAVA, don't know how to transform the model file(.caffemodel or .pb) from JAVA layer to Tengine. I know pure java for TFLite have a Interpreter class can do this.

Thanks

cyberfire, 2019-01-15:
@Lemiron24 
Maybe you can refer the android application example @ https://github.com/OAID/Tengine-app

closed 2019-01-15 by cyberfire