OAID/Tengine #78: caffe_wrapper issue: loop call of Classify will crashin tengine(classification.cpp)

houzh, 2018-12-11:
run tengine/examples/classification  .test OK
modify classification.cpp add a loop to call classify as follwing,tengine will caused crash
  CHECK(!img.empty()) << "Unable to decode image " << file;
  std::vector<Prediction> predictions;
  for(int i=0;i<3;i++)//add loop to call classify
     predictions = classifier.Classify(img);

cyberfire, 2018-12-12:
@houzh 
As prerun_graph () and postrun_graph() should only be called once, please check if prerun_graph()/postrun_graph() is inside your loop.

Thanks,

Haitao

cyberfire, 2019-01-15:
@houzh 
Is that possible for you to try the tengine 1.0 version? The caffe wrapper has been revised to call prerun/postrun only onece.

Thanks,

Haitao

closed 2019-03-02 by cyberfire