OAID/Tengine #85: mobilenet_ssd run time

vincent8877, 2018-12-16:
hi, thanks for the great work!

I make the example of mobilenet_ssd, and try to test the run time, but it seems the run time is instable and the function "get_input_data_ssd" cost too much time.

My paltform is firefly RK3399 and the stystem is ubuntu 16.04. I set the repeat time as 100 and test the run time in next 4 steps:

Step1: just the time of "set_tensor_buffer" and "run_graph"
[image]
[image]
[image]
as you can see, the first 22 repeats, the run time is very stable, but after that run time becomes instable

Step2: add the function "get_input_data_ssd"
[image]
[image]
[image]
the result looks alike as that of Step1, the first 11 repeats, time is stable and then instable, and after 100 repeats, the max_time and min_time varies a lot.
besides, compare to Step1 which avg_time is  158.798ms, the avg_time in Step2 is 232.255, means the function "get_input_data_ssd" takes 74ms per frame in average. I think this time is too long.

Step3: just the time of "get_input_data_ssd"
[image]
[image]
[image]
as the results show: in 0~12 repeats and 27~44 repeats, the run time is stable and acceptable, but in other repeats the time is instable and too long.

Step4: try to make the time of "get_input_data_ssd" shorter
[image]
[image]
[image]
in this Step, I modify the function  "get_input_data_ssd" ，just put the imread out of it, and test the time.
Although the result is still instable, the 0~2 repeats time seems very nice, and the avg_time (48.104ms) is shorter than that of Step3 (54.9964ms)

So here is my questions:
1. why the run time is instable, how can i do to make it stable?
2. the function "get_input_data_ssd" may take too long time i think, any suggestion to make it shorter?

Thank you.

hey-yahei, 2018-12-16:
Hi, vincent8877
Thanks for your careful experiment.
1. I think you may forget to set mode of gpu as **perfomance** before test([Heterogeneous Computing#Step 2](https://github.com/OAID/Tengine/wiki/GPU-CPU%E5%BC%82%E6%9E%84%E8%B0%83%E5%BA%A6%E8%B7%91MobilenetSSD)). What's more, processors will reduce their working frequency automatically when they detect the rise of temperature. To make it work more stably, you can take some measures such as cooling fins and cooling fans to reduce the temperature.
2. I'm sorry that `get_input_data_ssd` function is not accelerated by Tengine. If you need to make it shorter, try to parallel it by using OpenMP, pThread and so on. Using other lighter vision library instead of OpenCV may be an another option.

vincent8877, 2018-12-16:
thanks for reply @hey-yahei 
but actually i am not using GPU at the moment, i will try CPU/GPU running tomorrow
and the advices are instructive
thanks again

closed 2018-12-31 by cyberfire

renfeier, 2019-03-01:
> Hi, vincent8877
> Thanks for your careful experiment.
>
> 1. I think you may forget to set mode of gpu as **perfomance** before test([Heterogeneous Computing#Step [...]

so you think the reason of test time that instable  is the temperature?