OAID/Tengine #88: failed to set node ops for node: conv5_3_deconv op: Deconvolution

qiufengzhuoyu, 2018-12-26:
[图片]

cyberfire, 2018-12-31:
@qiufengzhuoyu 

please enable CONFIG_ARCH_BLAS=y in makefile.config, and have another try.

Thanks,

Haitao

dzyjjpy, 2019-04-23:
@qiufengzhuoyu @cyberfire @lyuchuny3 
Hi all, I met similar error **when run caffemodel with tengine in rk3399.**
when I excute the command, it shows:
WARNING: linker: /data/local/tmp/HeadPose: unsupported flags DT_FLAGS_1=0x8000001
ACL Graph Initialized
Driver: ACLGraph probed 1 devices
create graph success 
**failed to set node ops for node: ConvNdBackward202 op: Deconvolution**
/home/jiapy/workspace/Tengine-master/executor/lib/graph_task.cpp:229 failed to Prerun task on  dev executor: generic_probe
/home/jiapy/workspace/Tengine-master/executor/lib/graph_task.cpp:315 bad status: CREATED
--------------------------------------
repeat 1 times, avg time per run is 0.098 ms
 out_tensor point null

It will be appreciated if you can provide any useful advice. Thanks

cyberfire, 2019-05-04:
Deconvolution is implemented now.

closed 2019-05-04 by cyberfire