OAID/Tengine #89: Add sigmoid layer failed

jaysimon, 2018-12-28:
I am trying to add sigmoid layer, but failed, I got this:
```
./dnn5: symbol lookup error: /home/openailab/workspace/Tengine-master/build/libtengine.so: undefined symbol: _ZN7TEngine24RegisterSigmoid_NodeExecEv
```
I follow this [Operator Develop Guide
](https://github.com/OAID/Tengine/blob/master/doc/operator_dev.md) to add new layer
### 1. Add Operator Header File
- Define `sigmoid.hpp` to `operator/include/operator/sigmoid.hpp`
```cpp
#ifndef __SIGMOID_HPP__
#define __SIGMOID_HPP__
#include "operator.hpp"
namespace TEngine {
class Sigmoid: public OperatorNoParam<Sigmoid> {
public:
    Sigmoid() { name_="Sigmoid"; }
    Sigmoid(const Sigmoid&)= default;
    ~Sigmoid() {}
    void SetSchema(void) override;
};
} //namespace TEngine
#endif
```
### 2. Add Operator Cpp File
	- Set operator schema in file `operator/operator/sigmoid.cpp`
```cpp
#include "operator/sigmoid.hpp"
#include "static_graph.hpp"
namespace TEngine {
void Sigmoid::SetSchema(void)
{
    Input({"input:float32"})
   .Output({"output:float32"})
   .SetDoc(R"DOC(Sigmoid: only caffe flavor sigmoid)DOC");
}
} //namespace TEngine
```

Add `obj-y+=sigmoid.o` to `operator/operator/Makefile`

### 3. Add register sigmoid operator in file `operator/plugin/init.cpp`
```cpp
#include "operator/sigmoid.hpp"
RegisterOp<Sigmoid>("Sigmoid");
```
### 4. Add Implementation in Executor Folder
- Add `executor/operator/common/sigmoid.cpp`
```cpp
#include <iostream>
#include <functional>
#include <stdlib.h>

#include "logger.hpp"
#include "node_ops.hpp"
#include "tensor_mem.hpp"
#include "graph.hpp"
#include "operator/sigmoid.hpp"
#include <math.h>

namespace TEngine
{
namespace SigmoidImpl
{
struct SigmoidOps : public NodeOps
{
    bool Run(Node * node)
    {
        const Tensor * input_tensor=node->GetInputTensor(0);
        Tensor * output_tensor=node->GetOutputTensor(0);
        const TShape&  shape=input_tensor->GetShape();
        const std::vector<int> dims=shape.GetDim();
       int batch_number=dims[0];
        int channel_num=dims[1];
        int channel_size=dims[2]*dims[3];
        int img_size=channel_num*channel_size;
        
        const float * input=(const float *)get_tensor_mem(input_tensor);
        float * output=(float *)get_tensor_mem(output_tensor);

        for(int i=0;i<batch_number;i++)
        {
            for(int c=0;c<channel_num;c++)
            {
                int offset=i*img_size+c*channel_size;
                for(int l=0;l<channel_size;l++)
                {
                    output[offset+l]=1 / (1 + exp(-input[offset+l]));
                }
            }
        }
        return true;
    }
};
} //namespace SigmoidImpl

using namespace SigmoidImpl;
void RegisterSigmoid_NodeExec(void)
{
    SigmoidOps *ops = new SigmoidOps();
    NodeOpsRegistryManager::RegisterOPImplementor("common",
                                                  "Sigmoid", ops);
}
} //namespace TEngine
```
- Add `obj-y+=sigmoid.o` in file `executor/operator/common/Makefile`
### 5. Register Implementation in Executor Plugin
- Add `RegisterSigmoid_NodeExec` in file `executor/plugin/init.cpp`:
```cpp
extern void RegisterSigmoid_NodeExec(void);
RegisterSigmoid_NodeExec();
```
Any ideas?

hey-yahei, 2018-12-29:
Hi, jaysimon
Following [Operator Develop Guide](https://github.com/OAID/Tengine/blob/master/doc/operator_dev.md), you have created **Sigmoid** operation correctly. But note that you have not register operation for any serializer. 
For example, register it for caffe_serializer -- 
#### 1. Add load function for Sigmoid in Caffe Serializer     
Add `LoadCaffeSigmoid` in file [serializer/caffe/caffe_serializer.cpp](https://github.com/OAID/Tengine/blob/master/serializer/caffe/caffe_serializer.cpp):       
```cpp
static bool  LoadCaffeSigmoid(StaticGraph * graph, StaticNode * node, const te_caffe::LayerParameter& layer_param)
{
    StaticOp * op=CreateStaticOp(graph,"Sigmoid");

    SetNodeOp(node,op);

    return true;
}
```
#### 2. Register Sigmoid OP
Add following code in function `CaffeSerializerRegisterOpLoader` of file [serializer/caffe/caffe_serializer.cpp](https://github.com/OAID/Tengine/blob/master/serializer/caffe/caffe_serializer.cpp):
```cpp
p_caffe->RegisterOpLoadMethod("Sigmoid",op_load_t(LoadCaffeSigmoid));
p_buddy->RegisterOpLoadMethod("Sigmoid",op_load_t(LoadCaffeSigmoid));
```
#### 3. Rebuild Tengine and try again

jaysimon, 2019-01-03:
@hey-yahei  
Thanks for help! I follow your advicement, and it's solved.
One little thing, add #include sigmoid.hpp is aslo need, I added it in step 3.

1. Add load function for Sigmoid in Caffe Serializer
Add LoadCaffeSigmoid in file serializer/caffe/caffe_serializer.cpp:
```cpp
static bool  LoadCaffeSigmoid(StaticGraph * graph, StaticNode * node, const te_caffe::LayerParameter& layer_param)
{
    StaticOp * op=CreateStaticOp(graph,"Sigmoid");

    SetNodeOp(node,op);

    return true;
}
```
2. Register Sigmoid OP
Add following code in function CaffeSerializerRegisterOpLoader of file serializer/caffe/caffe_serializer.cpp:
```cpp
p_caffe->RegisterOpLoadMethod("Sigmoid",op_load_t(LoadCaffeSigmoid));
p_buddy->RegisterOpLoadMethod("Sigmoid",op_load_t(LoadCaffeSigmoid));
```
3. Add sigmoid.hpp
```cpp
#include "operator/sigmoid.hpp"
```
3. Rebuild Tengine and try again

cyberfire, 2019-01-10:
@jaysimon 

Glad to hear that your issue get resolved. We are going to create a demo to develop an  op in a standalone .so and the .so can be loaded by Tengine as plugin.

Thanks,

Haitao

closed 2019-01-18 by cyberfire

rahulsharma11, 2019-10-14:
Hi,
I am adding one "Elu" operator.
I followed above points for that. 

Please correct me if i am wrong, 
In point 5 above, i need to add 'RegisteryElu_NodeExec' in file "executor/operator/common/init.cpp"

> ### 5. Register Implementation in Executor Plugin
> * Add `RegisterSigmoid_NodeExec` in file `executor/plugin/init.cpp`:
> 
> ```c++
> extern void RegisterSigmoid_NodeExec(void);
> RegisterSigmoid_NodeExec();
> ```