OAID/Tengine #91: Is there any doc about supporting DLA

magicwyzh, 2019-01-09:
Hi, 
I've heard that the Tengine can support different hardware back-end(like DLA) as it provides HAL APIs. However, i didn't find docs about this part. 

I want to develop an FPGA(Xilinx/Intel programmable SoC) based DL application. There will be an ARM core and programmable logic that I can develop my own hardware accelerator for some compute-intensive operations. I wonder whether is it possible to use Tengine so that i can offload some computation to DLA while still leave some computations on the ARM core. 

Can anyone tell me where to find the doc about it?

Thanks:)

cyberfire, 2019-01-10:
@magicwyzh 

Here are the  framework to add a new DLA support in Tengine:

1.  Add new serializer model to load your own model
2.  Add new operator definition to represent the OPs defined by your DLA
3.  Add a new driver/device to execute the Ops new defined.

Thanks,

Haitao

gaojinwei01, 2019-01-10:
Hi Yizhi,

I'm interested in your project and would like to help. Please contact me through xiaowei@openailab.com.

Best regards,
Xiaowei

closed 2019-01-15 by cyberfire