OAID/Tengine #93: can not create the right staticgraph for pb file

feiliuzhixia1000m, 2019-01-10:
i have a tensorflow model, and i try to load it into tengine but failed. I debug the program and found that something worng in tf_serializer.cpp   MergeParentNode  function.

the function will erase parent node and insert the inputs of the parent node to the end of the base node's inputs, when the base node is conv2d  , the insert above will make input at index 1,and weight at index 0. So LoadConv2d fails. 
I change the MergeParentNode code to:
    auto input_ir = base_node->inputs.begin();
    unsigned int idx = 0;
    while(input_ir != base_node->inputs.end())
    {
        if(*input_ir == parent_node)
            break;

        input_ir++;
        idx ++;
    }

    base_node->inputs.erase(input_ir);

    /* connect parent's input node and base node */

    base_node->inputs.insert(base_node->inputs.begin()+idx, parent_node->inputs.begin(), parent_node->inputs.end());

   which will keep the order of Conv2d's inputs

cyberfire, 2019-01-10:
@feiliuzhixia1000m 

Thanks for debugging issue,

As the purpose of the merge  parents node is to cleanup some un-used nodes in tensorflow model, and it may happen that parent's node has more than one inputs,  so that the inputs of parent are placed in back. 
In order to resolve your issue, may be the policy could be optimized when the input of parent node just one, then do in-place replacement.

-Haitao

feiliuzhixia1000m, 2019-01-11:
@cyberfire 
May be the Merge will only be permitted when the node to be deleted has only one input,  or the two node have to be special, such as gemm without bias and a bias.

cyberfire, 2019-01-13:
@feiliuzhixia1000m 

That's a good suggestion, and we'll look into this.

-Haitao

closed 2020-09-12 by BUG1989