OAID/Tengine #94: load tensorflow model failed

feiliuzhixia1000m, 2019-01-11:
I tried to load a tensorflow model and failed . I debug the code and find in the tf_serializer.cpp : OptimizeGraph()
if(cur_node->op == "Reshape")
{
...
if(output_node->op == "Softmax" || output_node->op == "MatMul") 
...
under this circumstance if the input of current node  is "Const", the node("shape") will be deleted without save any reshape information. and the Reshape node will also be Merged without any reshape process .  
All these processes just delete the reshape operator violently.

Maybe no optimization will be better.

cyberfire, 2019-01-13:
@feiliuzhixia1000m 

Could you please share your model? As tensorflow is too flexible to compose ops, we have to deal with the cases one by one.

Thanks
Haitao

feiliuzhixia1000m, 2019-01-14:
sorry for not providing the  origin pb file, but i can describe my pb model

A is a input to Transpose node,with a transpose perm(const) , then Transpose's output is input into a Rashape node ,with a const  input "shape";
B is a Const to Taranpose_1 node ,with a transpose perm_1(Const), then Transpose_1's output is input into Reshape_1 node , with a Const input "shape_1";
Reshape node and Reshape_1 node are input to a MatMul node.
A Const Node(named "fc/dense/kernel") goes into a Identity Node(named "fc/dense/kernel/read"),and the goes into B.

In function OptimizeGraph(),firstly Const Shape and shape_1 are disconnected by the code:
               if(input_node0->op == "Const")
               {
                    DisconnectNode(input_node0);
                    input_node = input_node1;
                }
                else
                {
                    DisconnectNode(input_node1);
                    input_node = input_node0;
                } 
where the Param of Reshape get lost. Reshape and Reshape only has one input each.
Then in MergeChildNode(input_node, cur_node), Reshape  and Transpose, Reshape_1 and Transpose_1 are Merged just by move the input and output to Transpose(Transpose_1) node.

Then Transpose and Transpose_1 are input into MatMul, In LoadGemm function it requires one of it's input is Const, but either Transpose or Transpose_1 is Const .  So the code
StaticTensor* weight_tensor = FindTensor(graph, input1->name);
the dims.size() of weight_tensor is 0.

so segmentation default  in the code:
int k = weight_tensor->dims[0];

Maybe the next step will Merge the Const Node(named "fc/dense/kernel") and the  Identity Node(named "fc/dense/kernel/read") and the Transpose_1 Node, I'm no sure.

closed 2019-07-17 by cyberfire