OAID/Tengine #99: [Question] Model Zoo - Baidu

eymengunay, 2019-01-30:
Hello,

I recently stumbled upon this project and wanted to give it a try. However I realized that pretrained models are currently hosted on baidu which seems to ask for a Chinese mobile number to create an account.

Do you have any suggestions regarding how to download models?

Thank you!

vincent8877, 2019-01-31:
which model you want?
i can send to you by e-mail

eymengunay, 2019-01-31:
I'm mostly interested in testing the mobilenet_ssd implementation.

Thank you, I appreciate it!

closed 2019-01-31 by eymengunay

oanhrung, 2019-02-20:
@vincent8877 I also need the model mobilenet_ssd but I cannot download it. Can you send them to me by email? Thanks

eymengunay, 2019-02-20:
Hello @oanhrung ,

I can forward you the mail but unfortunately I can't see an email address on your profile.

Heimerdinger, 2019-02-25:
Hello @eymengunay ,
I also need the model to verify it. May you send them file to me by email? Many Thanks.

eymengunay, 2019-02-25:
Hello @Heimerdinger ,

I have just forwarded you the email.

Heimerdinger, 2019-02-25:
Hello @eymengunay ,
I receive it by mail. Very Thanks for your help.

pitchaya, 2019-04-04:
Would anyone be able to forward me mobilenet_ssd as well?  
   Thank you very much

eymengunay, 2019-04-04:
@pitchaya: You don't seem to have an email address on your profile (or it is not publicly visible)

Adhithya-tech, 2021-03-23:
@eymengunay  Could you forward mobilenet.tmfile to my email?

eymengunay, 2021-03-23:
Hello @Adhithya-tech,
I could but you have no information whatsoever on your profile. I have no clue what your email address is.

KrishnaShalini0518-arch, 2021-04-08:
Hello @eymengunay ,
I tried accessing tengine models through .deb file. But didn't find any.
Kindly share mobilenet_ssd.tmfile, retinaface.tmfile, yolact.tmfile to my email.

leeje2993, 2021-04-09:
Hello @eymengunay ,
Would you forward me mobilenet_ssd and darknet_yolo to my email?
Thank you in advance

eymengunay, 2021-04-10:
Hello @leeje2993,

Unfortunately I only have the mobilenet_ssd model which I just sent it to you.

KrishnaShalini0518-arch, 2021-04-10:
Hi @eymengunay ,

Can you share the mobilenet_ssd model to my email? Thank you very much.

eymengunay, 2021-04-10:
Oh hello @KrishnaShalini0518-arch,

I guess I overlooked your message and saw only the last one from @leeje2993. I have now forwarded it to you too.