OAID/Tengine #103: SSD object detection with Sigmoid Operator -> Segmentation fault

ghimiredhikura, 2019-02-16:
Hello, 

I used following reference to add Sigmoid layer operator 
1. https://github.com/OAID/Tengine/blob/master/doc/operator_dev.md
2. https://github.com/OAID/Tengine/issues/89

And rebuild the tengine in armv7l device (raspberrypi3b+) with `CONFIG_ARCH_BLAS=y`  
All the examples models provided in tengine are working fine but while I test with my own model it gives following error while calling function `run_graph(graph, 1);`. 
```ruby
Error: Segmentation fault 
```
Here is my caffemodel that I want to test: [mobilenet_v2_ssd_face.zip](https://github.com/OAID/Tengine/files/2871566/mobilenet_v2_ssd_face.zip)

Thanks for the help!

hey-yahei, 2019-02-24:
Hi, ghimiredhikura
I've add `Sigmoid` op as #89 and test your caffemodel in [examples/mobilenet_ssd](https://github.com/OAID/Tengine/tree/master/examples/mobilenet_ssd) on x86 with BLAS. It works well without errors as you show. I think you may set input size as 300x300 in your code, but your prototxt shows that the size of input is 320x320.
Thanks

ghimiredhikura, 2019-02-24:
Hello @hey-yahei, 

Did you test model that I have attached above with examples/mobilenet_ssd program? My model was trained for 320x320 and it works without any error in other platforms such as opencv/ncnn, but with tengine I get Segmentation fault error. Can you test in its original form, i.e., 320x320 size?

hey-yahei, 2019-02-24:
Hi, ghimiredhikura
Yes, I've test model that I have attached above with examples/mobilenet_ssd program. 
In the program, model is specified as the one you've attached above([L36&L37 in mssd.cpp](https://github.com/OAID/Tengine/blob/master/examples/mobilenet_ssd/mssd.cpp#L36)) and the size of input is setted as 320x320([L217&L218 in mssd.cpp](https://github.com/OAID/Tengine/blob/master/examples/mobilenet_ssd/mssd.cpp#L217)). It works without any errors here.

ghimiredhikura, 2019-03-07:
HI, @hey-yahei ,
What was your rpi os. In my case it works without error in aarch64 (armv8), but with armv7l (such as raspbian os) it gives same error. Segmentation fault.

goldlee, 2020-06-20:
@ghimiredhikura hey,guys,have you solved this problem?can you share with me?3q

closed 2020-09-12 by BUG1989