OAID/Tengine #104: "-p  4" is faster than  "TENGINE_CPU_LIST=4" ？

wangshankun, 2019-02-18:
RK3999 Ubuntu18.04
========================================== 
./build/tests/bin/bench_mobilenet **-p 4**
run-time library version: 1.0.0
REPEAT COUNT= 100
Repeat [100] time 103431.84 us per RUN. **used 10343184 us**
8.5976 - "n02123159 tiger cat"
7.9550 - "n02119022 red fox, Vulpes vulpes"
7.8679 - "n02119789 kit fox, Vulpes macrotis"
7.4274 - "n02113023 Pembroke, Pembroke Welsh corgi"
6.3646 - "n02123045 tabby, tabby cat"
ALL TEST DONE
==========================================
**export TENGINE_CPU_LIST=4**
./build/tests/bin/bench_mobilenet
ENV SET: [4]
run-time library version: 1.0.0
REPEAT COUNT= 100
Repeat [100] time 150165.59 us per RUN. **used 15016559 us**
8.5976 - "n02123159 tiger cat"
7.9550 - "n02119022 red fox, Vulpes vulpes"
7.8679 - "n02119789 kit fox, Vulpes macrotis"
7.4274 - "n02113023 Pembroke, Pembroke Welsh corgi"
6.3646 - "n02123045 tabby, tabby cat"
ALL TEST DONE

lyuchuny3, 2019-02-25:
Hi, I have tried on my fireflyRK3399 Ubuntu 16.04. For `-p 4`  or `export TENGINE_CPU_LIST`, it takes almost the same time to run bench_sqz: 
```
firefly@chun:~/Public/Tengine$ ./build/tests/bin/bench_sqz -p 4
run-time library version: 1.0.0-github
REPEAT COUNT= 100
Repeat [100] time 90703.75 us per RUN. used 9070375 us
0.2763 - "n02123045 tabby, tabby cat"
0.2673 - "n02123159 tiger cat"
0.1766 - "n02119789 kit fox, Vulpes macrotis"
0.0827 - "n02124075 Egyptian cat"
0.0777 - "n02085620 Chihuahua"
--------------------------------------
```

```
firefly@chun:~/Public/Tengine$ ./build/tests/bin/bench_sqz
ENV SET: [4]
run-time library version: 1.0.0-github
REPEAT COUNT= 100
Repeat [100] time 90341.36 us per RUN. used 9034136 us
0.2763 - "n02123045 tabby, tabby cat"
0.2673 - "n02123159 tiger cat"
0.1766 - "n02119789 kit fox, Vulpes macrotis"
0.0827 - "n02124075 Egyptian cat"
0.0777 - "n02085620 Chihuahua"
```

lyuchuny3, 2019-02-25:
Hi, 
Thank you for report this problem.
This is because `set_cpu_list` must set before `init_tengine`. I have updated the code. You can pull the newest codes and try. It should be solved now.

Thanks

closed 2019-02-25 by lyuchuny3