OAID/Tengine #110: arm平台下，1x1的卷积什么情况下conv_2d_fast，conv_2d_dw如何切换？

renfeier, 2019-03-01:
我的模型里面有1x1的卷积，当我运行sample，发现卷积跑的是conv_2d_Fast里面的run函数，是不是1x1的卷积用conv_2d_dw跑会更快呢？这2者的区别是啥？ @lyuchuny3

hey-yahei, 2019-03-01:
Hi, renfeier
`conv_2d_dw` is for depthwise-convolution while `conv_2d_fast` for common convolution. 
Note that depthwise-convolution is not equalvelant to 1x1 convolution. It's the 1x1 convolution with **groups == in_channels**.
Because the 1x1 conv in your model is not the common one, so Tengine just runs it with conv_2d_fast. 

Thanks

renfeier, 2019-03-04:
> Hi, renfeier
> `conv_2d_dw` is for depthwise-convolution while `conv_2d_fast` for common convolution.
> Note that depthwise-convolution is not equalvelant to 1x1 convolution. It's the 1x1 convolution with **groups == in_channels**.
> Because the 1x1 conv in your model is the common one, so Tengine just runs it with conv_2d_fast.
> 
> Thanks

I got it, Thanks a lot

closed 2019-03-04 by renfeier