OAID/Tengine #111: 现在，还没有支持YOLOV3吗

smartwell, 2019-03-01:
我在导入tiny－yolov3 模型时出错，
Message type "te_caffe.LayerParameter" has no field named "upsample_param".
upsample_param 这一层还没有支持

hey-yahei, 2019-03-01:
Hi, smartwell
I'm sorry that `upsample_param` operator is not surpported yet. Maybe you could try to implement it following [operator_dev](https://github.com/OAID/Tengine/blob/master/doc/operator_dev.md) and register it following [serializer_dev](https://github.com/OAID/Tengine/blob/master/doc/serializer_dev.md). 
Welcome everyone to contribute for Tengine and to make it stronger and stronger.

Thanks

doublegssc, 2019-06-21:
@hey-yahei 
您好～按照您在楼上的回复，我根据operator_dev的过程分别添加了upsample.hpp、upsample_param.hpp、upsample.cpp，并且在te_caffe.proto中添加了  optional UpsampleParameter upsample_param = 149;以及UpsampleParameter的定义。。在重新make工程（make通过，也生成了相应的.o文件）之后，导入tiny-yolov3模型还是报错：
[libprotobuf ERROR google/protobuf/text_format.cc:307] Error parsing text-format te_caffe.NetParameter: 511:20: Message type "te_caffe.LayerParameter" has no field named "upsample_param".
请问这种情况要怎么解决？
期待您的回复。。

gaojinwei01, 2019-08-24:
YoLo series models will be natively supported through darknet format in Tengine 1.9 version.

BUG1989, 2020-04-18:
@smartwell This model has been supported in latest version, please try it again~

closed 2020-04-18 by BUG1989

guoqiang0148666, 2020-04-26:
@gaojinwei01 @BUG1989 l run yolov3 meet segmantation fault. Here is prototxt file:

[Uploading yolov3.prototxt.txt…](yolov3.prototxt)

Besides, how to used darknet format yolov3 in Tengine

hufangjian, 2020-08-17:
I got the same error。

hufangjian, 2020-08-17:
> @gaojinwei01 @BUG1989 l run yolov3 meet segmantation fault. Here is prototxt file:
> 
> [Uploading yolov3.prototxt.txt…](yolov3.prototxt)
> 
> Besides, how to used darknet format yolov3 in Tengine
I got the sample error.

hufangjian, 2020-08-17:
> @gaojinwei01 @BUG1989 l run yolov3 meet segmantation fault. Here is prototxt file:
> 
> [Uploading yolov3.prototxt.txt…](yolov3.prototxt)
> 
> Besides, how to used darknet format yolov3 in Tengine
It may be help you

upgrade_net_proto_text  model_old.prototxt model_new.prototxt
upgrade_net_proto_binary model_old.caffemodel model.new.caffemodel