OAID/Tengine #112: Tengine编译失败：/usr/bin/ld: cannot find -lopenblas

chenjiaoAngel, 2019-03-01:
[image]
已经安装了liblapack-dev liblapack3 libopenblas-base libopenblas-dev
严格按照安装路径进行安装和编译的，链接地址：https://github.com/OAID/Tengine/blob/master/doc/install.md
makefile.config:
[image]

hey-yahei, 2019-03-01:
Hi, chenjiaoAngel
Please double check whether you've installed `libopenblas-dev` successfully? Does `libopenblas.so` file exist in `/usr/lib`? If yes, close your terminal and reopen the bash to try again.

Thanks

chenjiaoAngel, 2019-03-01:
`libopenblas-dev` has installed，and export LD_LIBRARY_PATH=/home/cj/anaconda2/envs/tengine_py2.7/lib/, but its doesn't work

hey-yahei, 2019-03-01:
Hi, chenjiaoAngel
Did you install `libopenblas-dev` by conda? 
What's more, try to run the command `ldconfig` to apply your new setting after export variable `LD_LIBRARY_PATH`.

Thanks

chenjiaoAngel, 2019-03-04:
hi, Thanks your advices.But it doesn't work, libopenblas is installed by conda. Do you have another ways to build tengine, for example using docker. I want to get some inference data of some public model on some android devices.

hey-yahei, 2019-03-04:
Hi, chenjiaoAngel
1. Please double check whether your openblas installed via conda is in `/home/cj/anaconda2/envs/tengine_py2.7/lib/` or not.
2. Run `export LD_LIBRARY_PATH=/home/cj/anaconda2/envs/tengine_py2.7/lib:$LD_LIBRARY_PATH` and then `ldconfig`.
3. If it doesn't work as well, try to 
    run `export LIBRARY_PATH=/home/cj/anaconda2/envs/tengine_py2.7/lib:$LIBRARY_PATH` and then `ldconfig`.
    ***Note that export variable `LIBRARY_PATH` but not `LD_LIBRARY_PATH` here.***
4. Maybe you can try to install openblas via `apt`(for ubuntu, debian, ....) or `yum`(for centos, fedora, ...) instead of conda.

feiliuzhixia1000m, 2019-03-04:
In Makefile,line 22
LD=$(CROSS_COMPILE)g++ $(SYSROOT_FLAGS) $(SYSROOT_LDFLAGS)

I think you need to add the path of openblas to  LD

chenjiaoAngel, 2019-03-05:
Thanks, I found the other way to build Tengine.

chenjiaoAngel, 2019-03-05:
I used the Tengine/doc/install.md to install tengine. When i complie tengine/examples, ocurred th problem: tengine/install/lib/libtengine.so: undefined reference to `cblas_sgemm'
buils and install in tengine don't meet problem

chenjiaoAngel, 2019-03-05:
[image]

feiliuzhixia1000m, 2019-03-05:
Maybe libopenblas is on your host ，so you can compile successfully,   but when you run you app on your target board, it cannot find libopenblas. So , install libopenblas on your target board and tell it the location

Good Luck

closed 2019-03-07 by chenjiaoAngel

knjwhn, 2020-03-04:
Hello,How can you solve this problem? I've met this problem:/home/henan/tengine/build/libhclcpu.so: undefined symbol: cblas_sgemm,Hope for your help.

guoqiang0148666, 2020-04-27:
@knjwhn 
In my x86_64 ubuntu platform. i modify CMakelist.txt (tengine root dir):
```
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -274,7 +274,9 @@ if(CONFIG_ARCH_BLAS)
        if(ANDROID)
                TARGET_LINK_LIBRARIES(hclcpu ${OPENBLAS_LIB_PATH}/libopenblas.so)
        else()
-               TARGET_LINK_LIBRARIES(hclcpu ${BLAS_LINKER_FLAGS})
+               TARGET_LINK_LIBRARIES(hclcpu ${OPENBLAS_LIB_PATH}/libopenblas.so)
+               #TARGET_LINK_LIBRARIES(hclcpu ${BLAS_LINKER_FLAGS})
        endif()
 endif()
``` 
Meanwhile assigning your OPENBLAS_LIB/INCLUDE_PATH in x86 config file.

if failed, you can also reference this [openblas](https://www.codelast.com/%E5%8E%9F%E5%88%9B-%E5%9C%A8%E6%A0%91%E8%8E%93%E6%B4%BE3%E4%B8%8A%E8%B7%91ell%E7%9A%84demo%E6%8A%A5%E9%94%99%EF%BC%9Aimporterror-build_darknetreference-so-undefined-symbol-cblas_sgemm/)

Good luck