OAID/Tengine #113: find a bug in serializer/tensorflow/tf_serialzier.cpp:LoadConv2D

feiliuzhixia1000m, 2019-03-04:
In function   LoadConv2D:
  if(GetAttrValue(node_def, "strides", value))
    {
        param.stride_h = value.list().i(1);
        param.stride_w = value.list().i(2);
    }
I think it shoud be changed to 
 if(GetAttrValue(node_def, "strides", value))
    {
        param.stride_h = value.list().i(2);
        param.stride_w = value.list().i(3);
    }

feiliuzhixia1000m, 2019-03-04:
and the same problem in the code:
if(GetAttrValue(node_def, "dilations", value))
    {
        param.dilation_h = value.list().i(1);  
        param.dilation_w = value.list().i(2); 
    }
I think it should be:
if(GetAttrValue(node_def, "dilations", value))
    {
        param.dilation_h = value.list().i(2);  
        param.dilation_w = value.list().i(3);  
    }

cyberfire, 2019-07-16:
@feiliuzhixia1000m 
According to https://tensorflow.google.cn/api_docs/python/tf/nn/conv2d, the strides definition is strides = [1, stride, stride, 1].

Thanks,

Haitao

closed 2019-07-17 by cyberfire