OAID/Tengine #114: Android prebuilt packages baidu problem

NikolaMr, 2019-03-06:
Hi!

I have recently stumbled upon Tengine, tested it on a linux machine and wanted to give it a try on my android device also following the https://github.com/OAID/Tengine/blob/master/doc/build_android.md. However, I have a problem with the provided baidu link https://pan.baidu.com/s/1RPHK_ji0LlL3ztjUa893Yg. I can't download from that link without username and I'm from Europe which kinda complicates things. Could you somehow share prebuilt packages by some other hosting service?

Best regards,
Nikola

lyuchuny3, 2019-03-08:
Hi, you can try this link: ftp://ftp.openailab.net.cn/Tengine_android_build/

NikolaMr, 2019-03-11:
@lyuchuny3 Thanks for the link it is working :)

closed 2019-03-13 by lyuchuny3