OAID/Tengine #115: Outputs are not the same between CPU and GPU.

hiigao, 2019-03-08:
I run input layer with single convolution layer on RK3399, and I find the outputs of this simple net on CPU and GPU are different.
The output shape is 24*150*150, the first 10 items from the output:
```
CPU Output: -42.7188  10.8438  86.9375  96.5  158  163.875  -70.1875  -107.375  -108.062  -119.75
GPU Output: -42.7404  10.7809  86.7634  96.2728 158.053  163.872  -70.4219  -106.988  -108.16 -119.889
```

The convolution layer are set to be a 7*7 convolution with pad=3, stride=2. Why they are so different?

closed 2019-03-08 by hiigao

pighead1016, 2019-03-12:
check the ”ACL_FP16“ is 1 or not，when you use ”gpu“ and ACL_FP16=1。 the float not the 4 bytes，just 16bit

rahulsharma11, 2019-10-15:
Hi,
Can you please tell me how to print the output shape of layers?
I want to print the shape of layers after model loaded (using onnx model)
Thanks.