OAID/Tengine #117: build on linux, make error

dzyjjpy, 2019-03-14:
When I installed Tengine on Linux follow the guide:(https://github.com/OAID/Tengine/blob/master/doc/install.md)
when $ make -j8, it shows:
 **fatal errors:  arm_neon.h, no such file or directory.**
**I had upacked  Tengine_android_build, can find the arm_neon.h in android-ndk-r16b**
How should I solve the issue? where should I add the include path? in CMakeList.txt or makefile.config?
Thanks.

hey-yahei, 2019-03-14:
Hi, dzyjjpy
Do you want to build Tengine for Android? If so, please follow the guide -- [build_android](https://github.com/OAID/Tengine/blob/master/doc/build_android.md).
Thanks

dzyjjpy, 2019-03-14:
I want to build it on Linux at first, then on Android. Both linux and android tutorials, I tried to build, but failed. Now it shows:
**error:#error "Neon support not enabled"**

hey-yahei, 2019-03-14:
Hi, dyzjjpy
Which platform do you want to build on? Neon is a part of arm-v8 processor.

If you want to build Tengine for x86, please install **openblas**, uncomment `CONFIG_ARCH_BLAS=y` and comment `CONFIG_ARCH_ARM64=y`.

If you want to build Tengine for arm(Linux) on x86, please follow [cross_compile](https://github.com/OAID/Tengine/blob/master/doc/cross_compile.md).

Thanks

dzyjjpy, 2019-03-14:
Ubuntu 14.04 with all the packages download required in this tutorials.
https://github.com/OAID/Tengine/blob/master/doc/build_android.md

dzyjjpy, 2019-03-19:
solved, caused by protocbuf, need to remove local pb clearly.

closed 2019-03-19 by dzyjjpy

erniuli, 2019-03-27:
@dzyjjpy 
你好，我碰到了和你类似的问题，首先按照linux build and Android build指导编译都正常通过了。
* 我想利用根目录下的CMakeList.txt自己构建，最终编译报executor/operator/common/lrn_arm.cpp:29:22: fatal error: arm_neon.h: No such file or directory
* 我的目的是想在x86下 用cmake构建，方便debug
* 目前，做了以下修改：根目录CMakeList.txt    option(CONFIG_ARCH_BLAS  "build blas  version" )OFF改为了ON，增加了set(PROTOBUF_DIR  ×× ）和add_subdirectory(examples) 
* 根据你的回答，PROTOBUF_DIR是需要设置为本地3.0.0的路径吗？