OAID/Tengine #118: How to enanble NEON Support?

dzyjjpy, 2019-03-14:
when building tengine(follow all the steps to edit and set paths ),  met the following error:
**error:#error "Neon support not enabled**
**Does anyone know how to enable NEON support?**

hey-yahei, 2019-03-14:
Hi, dzyjjpy
Which platform did you build Tengine on? Could you show me your `makefile.config`?
Thanks

dzyjjpy, 2019-03-14:
@hey-yahei makefile.config
# Set the target arch 
CONFIG_ARCH_ARM64=y

# Enable Compiling Optimization
CONFIG_OPT_CFLAGS = -O2

# uncomment to build the reference operators wrapped on Caffe
# CONFIG_CAFFE_REF=y

# Set the path of caffe used for compile and link,
# if caffe's operator calculation is enabled.
# CAFFE_ROOT = /home/firefly/caffe

# Enable GPU support by Arm Computing Library
# CONFIG_ACL_GPU=y
#

# Use BLAS as the operator implementation
#
# CONFIG_ARCH_BLAS=y

# Set the path of ACL 
# ACL_ROOT=/home/firefly/ComputeLibrary

# Enable other serializers 
CONFIG_CAFFE_SERIALIZER=y
# CONFIG_MXNET_SERIALIZER=y
# CONFIG_ONNX_SERIALIZER=y
# CONFIG_TF_SERIALIZER=y
CONFIG_TENGINE_SERIALIZER=y

# Enable Wrappers
# CONFIG_FRAMEWORK_WRAPPER=y

# version postfix
CONFIG_VERSION_POSTFIX=github

dzyjjpy, 2019-03-14:
# Set the target arch 
CONFIG_ARCH_ARM64=y

# Enable Compiling Optimization
CONFIG_OPT_CFLAGS = -O2

# uncomment to build the reference operators wrapped on Caffe
# CONFIG_CAFFE_REF=y

# Set the path of caffe used for compile and link,
# if caffe's operator calculation is enabled.
# CAFFE_ROOT = /home/firefly/caffe

# Enable GPU support by Arm Computing Library
# CONFIG_ACL_GPU=y
#

# Use BLAS as the operator implementation
#
# CONFIG_ARCH_BLAS=y

# Set the path of ACL 
# ACL_ROOT=/home/firefly/ComputeLibrary

# Enable other serializers 
CONFIG_CAFFE_SERIALIZER=y
# CONFIG_MXNET_SERIALIZER=y
# CONFIG_ONNX_SERIALIZER=y
# CONFIG_TF_SERIALIZER=y
CONFIG_TENGINE_SERIALIZER=y

# Enable Wrappers
# CONFIG_FRAMEWORK_WRAPPER=y

# version postfix

CONFIG_VERSION_POSTFIX=github

hey-yahei, 2019-03-14:
Hi, dzyjjpy
1. It seems that you build Tengine on x86 but set target arch as ARM64? If so, please uncomment `CONFIG_ARCH_BLAS=y` and comment `CONFIG_ARCH_ARM64=y`.
2. I notice that your makefile.config is for old version. Maybe your code is old and recloning from github is suggested.
Thanks

dzyjjpy, 2019-03-14:
Hi @hey-yahei ， I tried to uncomment CONFIG_ARCH_BLAS=y and comment CONFIG_ARCH_ARM64=y before. But it also had the issue:
arm_neon.h:31:2 error: #error You must enable NEON instructions(e.g. -mfloat-abi=neon....) to use arm_neon.h.
I download the latest version package, so it should be the newest makefile.config. I omit the comment part at the  beginning. Anyway, thanks.

dzyjjpy, 2019-03-15:
Does anyone have similar issue?

dzyjjpy, 2019-03-19:
issue caused by pb, solved now.

closed 2019-03-19 by dzyjjpy