OAID/Tengine #119: Build the example Error:  linker command failed with exit code 1

dzyjjpy, 2019-03-19:
when I build the examples in /tengine/example using the following command, it doesn't work.

#For armv8:
cp ~/ComputeLibrary/build_64/libarm_compute* ~/android-ndk-r16b/platforms/android-21/arch-arm64/usr/lib/
cd ~/tengine/example
mkdir build
cd build
 **../android_build_armv8.sh #** 
~/workspace/tengine/examples/build$ ../android_build_armv8.sh

#!/bin/bash
export ANDROID_NDK=/home/jiapy/workspace/tengine/Tengine_android_build/android-ndk-r16b
cmake -DCMAKE_TOOLCHAIN_FILE=$ANDROID_NDK/build/cmake/android.toolchain.cmake \
     -DANDROID_ABI="arm64-v8a" \
     -DANDROID_PLATFORM=android-21 \
     -DANDROID_STL=c++_shared \
     -DTENGINE_DIR=/home/jiapy/workspace/tengine \
     -DOpenCV_DIR=/home/jiapy/workspace/tengine/Tengine_android_build/opencv/sdk/native/jni\
     -DPROTOBUF_DIR=/home/jiapy/workspace/tengine/Tengine_android_build/protobuf_lib \
     -DBLAS_DIR=/home/jiapy/workspace/tengine/Tengine_android_build/Openblas020-android \
     ..
-- CMAKE_TOOLCHAIN_FILE = /home/jiapy/workspace/tengine/Tengine_android_build/android-ndk-r16b/build/cmake/android.toolchain.cmake
-- CMAKE_INSTALL_PREFIX = /usr/local
OpenCV_DIR dir /home/jiapy/workspace/tengine/Tengine_android_build/opencv/sdk/native/jni
list dir /home/jiapy/workspace/tengine/examples/.git
-- Configuring done
-- Generating done
-- **Build files have been written to: /home/jiapy/workspace/tengine/examples/build**
**android_build_armv8.sh content**

**make -j4     # errors in this step**
Errors information:
~/workspace/tengine/examples/build$ make -j8
[ 10%] Building CXX object yolov2/CMakeFiles/YOLOV2.dir/yolov2.cpp.o
[ 10%] Building CXX object faster_rcnn/CMakeFiles/FASTER_RCNN.dir/faster_rcnn.cpp.o
[ 10%] Building CXX object ssd/CMakeFiles/SSD.dir/ssd.cpp.o
[ 13%] Building CXX object tengine_model/convert/CMakeFiles/convert_caffe_to_tm.dir/convert_caffe_to_tm.cpp.o
[ 16%] Building CXX object mobilenet_ssd/CMakeFiles/MSSD.dir/mssd.cpp.o
[ 20%] Building CXX object lighten_cnn/CMakeFiles/LIGHTEN_CNN.dir/lighten_cnn.cpp.o
[ 23%] Building CXX object imagenet_classification/CMakeFiles/Classify.dir/classification.cpp.o
[ 26%] Building CXX object mtcnn/CMakeFiles/MTCNN.dir/mtcnn_utils.cpp.o
[ 30%] Building CXX object tengine_model/convert/CMakeFiles/convert_caffe_to_tm.dir/__/__/common/common.cpp.o
[ 33%] Building CXX object lighten_cnn/CMakeFiles/LIGHTEN_CNN.dir/__/common/common.cpp.o
[ 36%] Linking CXX executable convert_caffe_to_tm
[ 40%] Building CXX object mtcnn/CMakeFiles/MTCNN.dir/mtcnn.cpp.o
/home/jiapy/workspace/tengine/Tengine_android_build/Openblas020-android/arm64/lib/libopenblas.so: undefined reference to `stderr'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
tengine_model/convert/CMakeFiles/convert_caffe_to_tm.dir/build.make:100: recipe for target 'tengine_model/convert/convert_caffe_to_tm' failed
make[2]: *** [tengine_model/convert/convert_caffe_to_tm] Error 1
CMakeFiles/Makefile2:491: recipe for target 'tengine_model/convert/CMakeFiles/convert_caffe_to_tm.dir/all' failed
make[1]: *** [tengine_model/convert/CMakeFiles/convert_caffe_to_tm.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 43%] Building CXX object mtcnn/CMakeFiles/MTCNN.dir/test_mtcnn.cpp.o
[ 46%] Building CXX object ssd/CMakeFiles/SSD.dir/__/common/common.cpp.o
[ 50%] Building CXX object mobilenet_ssd/CMakeFiles/MSSD.dir/__/common/common.cpp.o
[ 53%] Building CXX object yolov2/CMakeFiles/YOLOV2.dir/__/common/common.cpp.o
[ 56%] Building CXX object faster_rcnn/CMakeFiles/FASTER_RCNN.dir/__/common/common.cpp.o
[ 60%] Linking CXX executable LIGHTEN_CNN
/home/jiapy/workspace/tengine/Tengine_android_build/Openblas020-android/arm64/lib/libopenblas.so: undefined reference to `stderr'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
lighten_cnn/CMakeFiles/LIGHTEN_CNN.dir/build.make:137: recipe for target 'lighten_cnn/LIGHTEN_CNN' failed
make[2]: *** [lighten_cnn/LIGHTEN_CNN] Error 1
CMakeFiles/Makefile2:305: recipe for target 'lighten_cnn/CMakeFiles/LIGHTEN_CNN.dir/all' failed
make[1]: *** [lighten_cnn/CMakeFiles/LIGHTEN_CNN.dir/all] Error 2
[ 63%] Building CXX object mtcnn/CMakeFiles/MTCNN.dir/__/common/common.cpp.o
[ 66%] Building CXX object imagenet_classification/CMakeFiles/Classify.dir/model_config.cpp.o
[ 70%] Linking CXX executable SSD
[ 73%] Linking CXX executable MSSD
[ 76%] Linking CXX executable YOLOV2
[ 80%] Linking CXX executable FASTER_RCNN
[ 83%] Building CXX object imagenet_classification/CMakeFiles/Classify.dir/__/common/common.cpp.o
[ 86%] Linking CXX executable MTCNN
[ 90%] Linking CXX executable Classify
/home/jiapy/workspace/tengine/Tengine_android_build/Openblas020-android/arm64/lib/libopenblas.so: undefined reference to `stderr'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
mobilenet_ssd/CMakeFiles/MSSD.dir/build.make:137: recipe for target 'mobilenet_ssd/MSSD' failed
make[2]: *** [mobilenet_ssd/MSSD] Error 1
CMakeFiles/Makefile2:415: recipe for target 'mobilenet_ssd/CMakeFiles/MSSD.dir/all' failed
make[1]: *** [mobilenet_ssd/CMakeFiles/MSSD.dir/all] Error 2
/home/jiapy/workspace/tengine/Tengine_android_build/Openblas020-android/arm64/lib/libopenblas.so: undefined reference to `stderr'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
ssd/CMakeFiles/SSD.dir/build.make:137: recipe for target 'ssd/SSD' failed
make[2]: *** [ssd/SSD] Error 1
CMakeFiles/Makefile2:195: recipe for target 'ssd/CMakeFiles/SSD.dir/all' failed
make[1]: *** [ssd/CMakeFiles/SSD.dir/all] Error 2
/home/jiapy/workspace/tengine/Tengine_android_build/Openblas020-android/arm64/lib/libopenblas.so: undefined reference to `stderr'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
yolov2/CMakeFiles/YOLOV2.dir/build.make:137: recipe for target 'yolov2/YOLOV2' failed
make[2]: *** [yolov2/YOLOV2] Error 1
CMakeFiles/Makefile2:140: recipe for target 'yolov2/CMakeFiles/YOLOV2.dir/all' failed
make[1]: *** [yolov2/CMakeFiles/YOLOV2.dir/all] Error 2
/home/jiapy/workspace/tengine/Tengine_android_build/Openblas020-android/arm64/lib/libopenblas.so: undefined reference to `stderr'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
faster_rcnn/CMakeFiles/FASTER_RCNN.dir/build.make:137: recipe for target 'faster_rcnn/FASTER_RCNN' failed
make[2]: *** [faster_rcnn/FASTER_RCNN] Error 1
CMakeFiles/Makefile2:250: recipe for target 'faster_rcnn/CMakeFiles/FASTER_RCNN.dir/all' failed
make[1]: *** [faster_rcnn/CMakeFiles/FASTER_RCNN.dir/all] Error 2
/home/jiapy/workspace/tengine/Tengine_android_build/Openblas020-android/arm64/lib/libopenblas.so: undefined reference to `stderr'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
mtcnn/CMakeFiles/MTCNN.dir/build.make:167: recipe for target 'mtcnn/MTCNN' failed
make[2]: *** [mtcnn/MTCNN] Error 1
CMakeFiles/Makefile2:85: recipe for target 'mtcnn/CMakeFiles/MTCNN.dir/all' failed
make[1]: *** [mtcnn/CMakeFiles/MTCNN.dir/all] Error 2
/home/jiapy/workspace/tengine/Tengine_android_build/Openblas020-android/arm64/lib/libopenblas.so: undefined reference to `stderr'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
imagenet_classification/CMakeFiles/Classify.dir/build.make:152: recipe for target 'imagenet_classification/Classify' failed
make[2]: *** [imagenet_classification/Classify] Error 1
CMakeFiles/Makefile2:360: recipe for target 'imagenet_classification/CMakeFiles/Classify.dir/all' failed
make[1]: *** [imagenet_classification/CMakeFiles/Classify.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

closed 2019-03-20 by dzyjjpy

pangchaogit, 2019-03-25:
I got same problems, how do you deal with it?

dzyjjpy, 2019-03-25:
change api level from 21 to 26, I solved the issue

dzyjjpy, 2019-03-25:
@pangchaogit

pangchaogit, 2019-03-26:
Thank you! , I fixed the problem by your ways.